pylessard/python-can-isotp #2: OSError: [Errno 93] Protocol not supported

MarthaDora, 2018-09-19:
Hi, I use this project on RasberryPi3, I used the example code to test,and I got an error when I run `s = isotp.socket()` , It confused me, actually. 
Error:
`  File "/usr/local/opt/python-3.7.0/lib/python3.7/socket.py", line 151, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 93] Protocol not supported
`

ppistorius, 2018-09-24:
Hi MarthaDora,

I think you need to compile and load the isotp kernel module like described here:
https://github.com/hartkopp/can-isotp-modules

The linux kernel supports can, but not isotp. But with loading the isotp kernel module, you are able to use isotp.

Best Regards
Paul

MarthaDora, 2018-09-25:
> Best Regards

Thank you for your advice，I asked a silly question，This problem is mentioned in Troubleshooting.

closed 2018-09-25 by MarthaDora