pylessard/python-can-isotp #4: Use Extended Adresses

HyDz, 2018-12-17:
Hi,
Thanks for your work.
Can you share an example using extended addressing?
When I use 8 char addresses, it send only the 3 last. I've tried to find the good options but I probably miss something.

Regards

hartkopp, 2018-12-17:
Hi,
the check is pretty simple: When the value for the can_id  is given with 8 chars you need to OR the value with CAN_EFF_FLAG, e.g. in
https://github.com/linux-can/can-utils/blob/master/isotpsend.c#L96
Did this answer your question?
Another example: https://github.com/linux-can/can-utils/blob/master/candump.c#L504
The idea of CAN filters is described here:
https://elixir.bootlin.com/linux/v4.9/source/Documentation/networking/can.txt#L446
And you can play e.g. with <candump> to get behind the filter behavior. When CAN_EFF_FLAG is set in can_mask this bit becomes relevant for filtering too.

HyDz, 2018-12-17:
Hi,
First thanks for your fast response, and your work on isotp.
I now use:
my_socket.bind("can0", rxid=0x1A2B3C4D | socket.CAN_EFF_FLAG, txid=0xA1B2C3D4 | socket.CAN_EFF_FLAG)
instead of
my_socket.bind("can0", rxid=0x1A2B3C4D, txid=0xA1B2C3D4)
And It works as expected. 
Thanks a lot

pylessard, 2018-12-17:
Reopening this issue. this should've worked.
@HyDz what version of the module are you using?

@hartkopp I took note of your suggestion in #3 and will review soon. I am sure you are right. I am actually restructuring this repo as it will soon offer more than just syntax candy, so everything should be fixed/release together. I will make sure to fix this and document everything.

Cheers

HyDz, 2018-12-18:
Hi,
Sorry for the delay, I am in France.
The version I used is the 0.4 installed with pip.