pylessard/python-can-isotp #5: Support for ISOTP extended adressing

polybassa, 2018-12-19:
Hi @pylessard,
great work!
I checked your code, since I'm always interested in different approaches for ISOTP implementations. 

I missed one feature of your code. Some OEMs use extended ISOTP addressing for various purposes. 
I don't talk about extended CAN identifiers. The ISOTP standard supports extended addresses in the first byte of the CAN data frame. 
You can generate this extended addressing for example with isotpsend 
```
echo "de ad be ef" | isotpsend -s 641 -d 251 -x 41:51 vcan0
```

pylessard, 2018-12-19:
Ah! you got me. 
I was hesitating about supporting other addressing mode as I've never dealt with an OEM that used something else than Normal fixed addressing. I thought that supporting these would be useless effort, but you just gave me an argument against it :)  Maybe @hartkopp could also share his experience/opinion on this.

What about Mixed addressing, do you think they're relevant?

polybassa, 2018-12-19:
Could you specify, what do you mean with normal fixed / mixed addressing? 
I observed two ways, how extended addressing is used, in the wild. 
GM uses the first byte of a CAN frame to indicate broadcast messages. So if you want to send a GMLAN broadcast message, it's useful, if you can specify a extended address for TX. 
BMW uses TX and RX extended addresses. They kind of encode the sender address of a message into the first byte of a CAN frame. This address is used in a response, to compute the CAN identifier on a response message. This means, `source can_id`, `destination can_id`, `extended address tx` and `extended address rx` are related to each other. Example: 
```
echo "de ad be ef" | isotpsend -s 640 -d 6FE -x 40:FE vcan0
```

polybassa, 2018-12-19:
Ah ok, thanks for this clarification. I only discovered the addressing scheme, I described.

hartkopp, 2018-12-19:
> Maybe @hartkopp could also share his experience/opinion on this.

I think @polybassa already summarized some use-cases that also make sense to me. I do not remember the details when I was implementing the extended addressing feature, but it was hard to specify the sockopts for it - and offer the addressing options for isotpsend/isotprecv . You get crazy with all the possible rx and tx relations  :-)
At least the ISO spec requires this address scheme - so I supported it.

pylessard, 2018-12-19:
> BMW uses TX and RX extended addresses. They kind of encode the sender address of a message into the first byte of a CAN frame. This address is used in a response, to compute the CAN identifier on a response message. 

Right, I've did a project for one of their vehicle and I remember that they were transmitting an address on the first byte of a CAN message. This was a way to multiplex the CAN ID as their CAN matrix are huge. What I saw was for every CAN message, not for IsoTP transport protocol though. 

Anyway, I'll check into supporting other addressing mode and try to find a way to make this user-friendly

pylessard, 2019-01-02:
@polybassa V1.3 is officially released, supporting all addressing mode. Documentation is there also.

Feel free to comment!

polybassa, 2019-01-03:
Ok, thanks for this information. I'll check your implementation, as soon as I find some free time.