pylessard/python-can-isotp #7: How can I send long CAN message greater than 8 bytes in python-can-isotp ?

Prriyamvradha, 2019-03-05:
Hi, I am new to python. I am working on a project using CANtact toolkit I am using two toolkits one to send the CAN message and others to receive the CAN message. I created a socket between the two devices and it worked fine for sending and receiving the standard CAN message of 8 Byte. Now I want to send more than 8 bytes of the message and I came across can-isotp protocol. I have used python-can for implementation of my project so far so I would like to continue with python-can-isotp. What is the simple code to send and receive CAN message using python-can-isotp? Also, should I create a new socket to receive more than 8 bytes of data? I have used the following code to create a socket between two CANtact devices. What change has to be made to the code to make it accept more than 8 bytes of data?I want to send and receive a CAN message like this [127, 51, 17, 124, 242, 102, 165, 37, 160, 41, 165, 149, 65, 179, 232, 15, 63, 76, 4, 0, 90, 72, 203, 250, 155, 188, 169, 58, 114, 16, 85, 107]

```
#!/bin/env python
import socket
import struct

canformat = '<IB3x8s'

class CanBridge():
    def __init__(self, interface_from, interface_to):
        self.interface_from = interface_from
        self.interface_to = interface_to
        self.canSocket_to = socket.socket(socket.PF_CAN, socket.SOCK_RAW, socket.CAN_RAW)
        self.canSocket_from = socket.socket(socket.PF_CAN, socket.SOCK_RAW, socket.CAN_RAW)
        try: 
            self.canSocket_to.bind((self.interface_to,))
            self.canSocket_from.bind((self.interface_from,))
        except OSError: 
            print("Could not bind to interfaces")
        #put the sockets in blocking mode.
        self.canSocket_to.settimeout(None)
        self.canSocket_from.settimeout(None)

    def run(self):
        while True:
            raw_bytes = self.canSocket_from.recv(4096)#512
            try:
                self.canSocket_to.send(raw_bytes)
            except OSError: #Buffer overflow usually from lack of connection.
                pass

            rawID,DLC,candata = struct.unpack(canformat,raw_bytes)
            canID = rawID & 0x1FFFFFFF
            candata_string = " ".join(["{:02X}".format(b) for b in candata])
            print("{:08X} {}".format(canID, candata_string))

if __name__ == '__main__':
    bridge = CanBridge('slcan1','slcan0')
    bridge.run()
```

pylessard, 2019-03-05:
Hi @Prriyamvradha 
Thanks for your interest. Most of your questions can be answered by reading the python-can-isotp documentation and socketcan documentation. Also we normally try to keep issues to track improvements, a site such as stack overflow might be the best place to ask your questions.  

This being said, here's what you need to understand. When communicating with your device through a socket, you make usage of SocketCan, a kernel module inside your OS that has the device driver included; the socket is just the interface for communication that Linux knows. You could have used your device by talking to the driver directly (without the use of socket).  

To use a CAN device in Python, you can use [python-can](https://python-can.readthedocs.io). It makes the usage of different device brands quite uniform and also supports some hardware that SocketCAN does not support. Note that python-can is able to talk with SocketCan as well. I'd say it is better to use python-can in Python than raw SocketCAN as you can easily switch between SocketCan and direct access to drivers.

Now, to use IsoTP protocol, you have few options.  You can use [this linux module](https://github.com/hartkopp/can-isotp) that implements IsoTP in C++. It extends SocketCAN and works with socket as well. Note that if you want to use it in Python, you'll need Python 3.7+ otherwise, you won't be able to create the socket by passing "CAN_ISOTP" for the protocol parameter. What you need to do is compile the module, load it in your OS then create a socket and use it.

This project offers you 2 possibilities : 
 1. An simplified socket interface for [hartkopp/can-isotp device driver](https://github.com/hartkopp/can-isotp) (because it is a little complicated to configure the protocol without it). See the `isotp.socket` object
 2. An implementation of the protocol directly in Python (in the user space). This implementation can be interfaced with python-can.

Note that the device driver (option #1) has better timing respect, while the user space implementation (option #2)  is more flexible : can be coupled with any hardware usable in Python and works under Windows as well.

Look at the example section of the documentation. You'll find some for the [Socket option](https://can-isotp.readthedocs.io/en/latest/isotp/socket.html) and some for the [PythonCan compatible implementation](https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can)

Regards

Prriyamvradha, 2019-03-05:
Thank you so much for the valuable input.

closed 2019-03-05 by Prriyamvradha