pylessard/python-can-isotp #9: CAN FD long message support

markuspi, 2019-04-03:
CAN-FD allows messages longer than 8 bytes. When setting `ll_data_length` to a value larger than 8 and then sending data longer than 8 bytes, python-can-isotp is encoding the length paramter incorrectly. In some cases, the length is even overflowing in the high bits of the first byte and thus changing the frame type. Receiving long messages is probably affected as well.

### Reproduction
```python
def show_msg(msg):
    print("can msg:", msg.data.hex())
    print()

address = isotp.Address(rxid=0x123, txid=0x456)
tl = isotp.TransportLayer(rxfn=lambda: None, txfn=show_msg, address=address, params={
    'll_data_length': 32, 
    'tx_padding': 0x00
})

for l in (5, 10, 20):
    data = bytearray(list(range(0xA1, 0xA1+l)))
    print("len:", l, "data:", data.hex())
    tl.send(data)
    tl.process()
```

#### Current Output
```
len: 5 data: a1a2a3a4a5
can msg: 05a1a2a3a4a50000000000000000000000000000000000000000000000000000

len: 10 data: a1a2a3a4a5a6a7a8a9aa
can msg: 0aa1a2a3a4a5a6a7a8a9aa000000000000000000000000000000000000000000

len: 20 data: a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4
can msg: 14a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b40000000000000000000000
```

#### Expected Output
According to ISO 15765-2 Section 9.6, SF_DL should be stored in the 2nd Byte if CAN_DL exceeds 8 bytes. Bits 0-3 of the first byte should be set to zero:
```
len: 5 data: a1a2a3a4a5
can msg: 05a1a2a3a4a50000000000000000000000000000000000000000000000000000

len: 10 data: a1a2a3a4a5a6a7a8a9aa
can msg: 000aa1a2a3a4a5a6a7a8a9aa0000000000000000000000000000000000000000

len: 20 data: a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b4
can msg: 0014a1a2a3a4a5a6a7a8a9aaabacadaeafb0b1b2b3b400000000000000000000

```

pylessard, 2019-04-03:
Thank you for this!
I will review soon.

pylessard, 2019-04-04:
What version of the standard are you referring to?  V2004 and V2011 doesn't have a section 9.6.

markuspi, 2019-04-04:
It's ISO 15765-2:2016. As far as I know, CAN FD has been released in 2012

hartkopp, 2019-04-04:
Not really. There first was a whitepaper from Bosch shown at the CAN-in-Automation conference (iCC) in 2012. In Summer 2012 we had a general CAN FD support in the Linux 3.6 kernel - but only with virtual CAN interfaces. Since Linux 3.15 real CAN FD was supported. The CAN FD for ISO15765-2 was developed in 2013/2014 and finally lead to ISO 15765-2:2016.
Btw. the remark from @markuspi regarding the SF_DL escape sequence is completely right.
https://github.com/hartkopp/can-isotp-modules/commit/44e7c2e457ced3f5d37c30af03f0a5b0430a4d31#diff-dc5186c73dcf461cd2e131a4e5f8e551R573

pylessard, 2019-04-05:
I just put my hand on the 2016 version. Will fix next week or the one after.
Regards

pylessard, 2019-04-27:
Sorry for the delay. 
I think it is correctly supported now. Will make a new release soon. I am new to CAN FD, so I did according to the best of my understanding. 

I added few things

- Support for long Single Frame.
- Restricted values of ll_data_length to 8,16,20,24,32,48,64
- Support for frame longer than 4095 (escape sequence on a FirstFrame)

I'd be happy to get feedback from somebody who is knowledgable with CAN FD.

Cheers

closed 2019-04-27 by pylessard

pylessard, 2019-05-01:
Fix release in v1.4

Thank you for the good quality bug report.

commit 3be86735 (2019-04-27): Correctly supports CAN FD escape sequence for Single Frame. Fixes #9