pylessard/python-can-isotp #13: UDS client cannot receive segmented frames over CAN-FD when ll_data_length does not match UDS server configuration

cpavot, 2019-06-24:
In the current implementation, CAN frame are conveyed on a CAN-FD bus instead of CAN-FD frames. The following pull request proposes a fix for this: https://github.com/pylessard/python-can-isotp/pull/12. It is necessary to include this fix to reproduce the below issue.

Using the above pull request, a UDS client is able to send segmented CAN-FD frames to a UDS server. However, a UDS client is not always able to properly receive segmented CAN-FD frames. It depends on the ll_data_length configuration parameter.

The behaviour of the python-can-isotp layer has been compared with kernel module implementation: https://github.com/hartkopp/can-isotp. The kernel module works fine with different configuration (note the ll_data_length parameter is named tx_dl in this module - the name explicitly specifies that it applies to Tx only).

Below is a table listing the results for some testing: KS = Kernel Space (hartkopp's module), US = User Space (python-can-isotp).

Test #|UDS server isotp layer|UDS server ll_data_length|UDS client isotp layer|UDS client ll_data_length|Reception of CAN-FD segmented frame
------|----------------------|-------------------------|----------------------|-------------------------|-----------------------------------
1|US|64|US|64|OK
2|US|8|US|8|OK
3|US|8|US|64|NOK: Received invalid CAN frame
4|US|64|US|8|NOK: Reception of CONSECUTIVE FRAME timed out
5|KS|64|KS|64|OK
6|KS|8|KS|8|OK
7|KS|8|KS|64|OK
8|KS|64|KS|8|OK
9|KS|8|US|64|NOK: Received invalid CAN frame
10|KS|64|US|8|NOK: Reception of CONSECUTIVE FRAME timed out

From these tests, we can observe the following:

1. If UDS client's ll_data_length parameter matches UDS server's ll_data_length, the segmented CAN-FD frames are properly received. 

1. If UDS client's ll_data_length parameter is less than UDS server's ll_data_length, python-can-isotp layer raises a "Reception of CONSECUTIVE FRAME timed out" exception:  https://github.com/pylessard/python-can-isotp/blob/f30f9754a5ed477085c5d12e3960df3d27b8347e/isotp/protocol.py#L437

1. If UDS client's ll_data_length parameter is larger than UDS server's ll_data_length, python-can-isotp layer raises a "Received invalid CAN frame: First frame specifies a length that is inconsistent with underlying CAN message DLC" exception: https://github.com/pylessard/python-can-isotp/blob/f30f9754a5ed477085c5d12e3960df3d27b8347e/isotp/protocol.py#L112

It seems that the implementation makes the assumption that UDS server shall send segmented frames according to UDS client's ll_data_length configuration - which is likely wrong. My understanding is that there is no negociation regarding this parameter. UDS client ll_data_length parameter shall only apply to transmitted messages. UDS client shall be able to process different data lengths and not rely on its own configuration.

To reproduce:
- Make sure to use a CAN-FD bus + the above fix to convey CAN-FD frames
- Send an UDS request expecting a long frame as an answer (i.e. requiring segmentation) to an ECU
- Play with ll_data_length on UDS client / server sides and try the 3 above use cases.

cpavot, 2019-06-24:
**Log for test 3**
''' (2019-06-24 18:17:40.083189)  UDS client  [04]  03 22 FD 00               '."..'
 (2019-06-24 18:17:40.174677)  UDS server  [08]  11 03 62 FD 00 00 01 02   '..b.....'
'''

pylessard, 2019-06-24:
Interesting. Thank you for the detailed bug report.  I have indeed not tested much with CAN FD. Will review soon

cpavot, 2019-06-24:
Ok, thanks! Let me know if you need additional data, I can easily reproduce.

pylessard, 2019-06-25:
Hi, I've reviewed and I acknowledge the issue. ``ll_data_length`` is used both for reception and transmission.  Reception length should be determined throught the DLC.

This will require some time, I will try to not make it too long. In the mean time, I am sure you have some script to test the above test case. Why not share them or try to include them in the unit test suite?  There is a file named ``test_layer_against_socket.py`` ready to greet new test cases :)

cpavot, 2019-06-27:
Hi, I heard you. :)
See https://github.com/pylessard/python-can-isotp/pull/14.

pylessard, 2019-07-02:
I have a working candidate [here](https://github.com/pylessard/python-can-isotp/commits/can_fd_fix), Reception does not look at ``tx_data_length`` anymore. ``len(msg.data)`` is used to define RX_DL.
I did few modifications:

- Parameter is called ``can_fd`` instead of ``is_fd``. I wanted to avoid confusion with the CanMessage property.
- I changed the parameter ``ll_data_length`` name for ``tx_data_length``. Former name is still supported for backward compatibility.
- I modified your test suite proposed in #14 in few ways
  - Renamed variables to match new parameter names
  - Renamed some functions, just to make things a little clearer.
  - Set default ``fd`` parameter to False, then manually set to True in your test. Otherwise it was breaking existing test when MTU of vcan0 is 16.
  - Gracefully skip your test when interface MTU is 16 instead of 72.
- Some misusage that forbidden by the standard and were previously tolerated (such as single frame with length > 8 and no escape sequence) are now striclty forbidden. Fixing this issue kind of highligted that compliance with ISO-15765-2 had to be improved.

Finally, fixing this brought me to change some stuff in the core of the TranportLayer; good news is that it made it more simple :) 

I'll let you have a look, will merge with master once I get your feedback.

P.S. Don't look too much into intermediate commits. It took me few trial to finally head in the right direction :)

cpavot, 2019-07-02:
Hi, thanks for the patches! I had a look but did no go into the details of the implementation. They seems OK. I did test on my setup and the issues are likely fixed. Thanks again.

pylessard, 2019-07-02:
Perfect. Will double check to doc tonight, merge with amster and make a new release.
Thanks for your help on this.

cpavot, 2019-07-02:
I forgot to say that I put a small comment about an ambiguous statement in the doc, but that's minor.

pylessard, 2019-07-03:
Fixed in v1.5
Thank you again.

closed 2019-07-03 by pylessard