pylessard/python-can-isotp #15: UDS client gets a timeout when receiving segmented frames while the blocksize within isotp_params is set to zero

VyshakhPV, 2019-06-27:
Hi,

I am using the udsoncan along with python-can-isotp and python-can for communication with my server.  I have set the blocksize for isotp to be zero which should indicate an infinitely large block size. 

But when I try using read_data_by_identifier to get data from the server, the reply for which will be send in more than two segmented frames , an "integer division or modulo by zero" error is also printed out and a TimeoutException is triggered. But I have verified that the server is responding to the request properly.

The log during the same is a provided below

> 2019-06-27 13:41:42 [ERROR] Connection: integer division or modulo by zero
> 2019-06-27 13:41:43 [DEBUG] Connection: No data received: [TimeoutException] - Did not receive frame from user queue in time (timeout=1 sec)
> 2019-06-27 13:41:43 [ERROR] UdsClient: [TimeoutException] : Did not receive response in time. P2 timeout time has expired (timeout=1.000 sec)

If I increase the blocksize more than zero, the data is properly received. 

Isn't the above mentioned behaviour a bug ?

pylessard, 2019-06-27:
Issue has been transerred to the can-isotp module. 
Problems seems to be caused by :   
https://github.com/pylessard/python-can-isotp/blob/442322902889e88baf2f3251b30028f2361085df/isotp/protocol.py#L496

pylessard, 2019-06-28:
Hi, this issue is now fixed. I am waiting to finish handling #13 before making a release.
I was sure there was a unit test for this scenario, but it appears that there was none! Glad you found this out.

closed 2019-06-28 by pylessard

VyshakhPV, 2019-06-28:
Hi,
It is great news and I am glad that I could be helpful in this

pylessard, 2019-07-03:
Fixed in v1.5

commit 51285cf9 (2019-06-28): Fix #15. Blocksize of 0 is now correctly supported.