pylessard/python-can-isotp #17: Overflow flow status is ignored

iwasz, 2019-11-26:
I think your library should call error_handler in such a case. My setup : using your library I'm sending a message like this :

```python
#!/usr/bin/env python3

import isotp
import logging
import time
import can
import msgpack

def my_error_handler(error):
    logging.warning('IsoTp error happened : %s - %s' %
                    (error.__class__.__name__, str(error)))

bus = can.interface.Bus(bustype='socketcan', channel='slcan0', bitrate=500000)

addr = isotp.Address(isotp.AddressingMode.NormalFixed_29bits,
                     source_address=0x11, target_address=0x22)

stack = isotp.CanStack(bus, address=addr, error_handler=my_error_handler)

# Just some data, not relevant
stack.send(msgpack.packb({"sensor": "accel", "time": 1,
                          "data": [1.2, 3.4]}, use_bin_type=True))

while stack.transmitting():
    stack.process()
    time.sleep(stack.sleep_time())

print("Payload transmission done.")

bus.shutdown()
```

Now another library running on a microcontroller (https://github.com/iwasz/cpp-can-isotp) is listening and it responds with a flow frame, and everything works if the message is short enough. Below I attach the ```candump any``` output :

```
  slcan0  18DA2211   [8]  10 2C 83 A6 73 65 6E 73
  slcan0  18DA1122   [3]  30 00 0A <- 3 == flow frame, 0 == CONTINUE TO SEND
  slcan0  18DA2211   [8]  21 6F 72 A5 61 63 63 65
  slcan0  18DA2211   [8]  22 6C A4 74 69 6D 65 01
  slcan0  18DA2211   [8]  23 A4 64 61 74 61 92 CB
  slcan0  18DA2211   [8]  24 3F F3 33 33 33 33 33
  slcan0  18DA2211   [8]  25 33 CB 40 0B 33 33 33
  slcan0  18DA2211   [4]  26 33 33 33
```

But if I try to send too much data (where too much means more than 128, because listening peer is constrained to this number), and the listening peer responds with ```overflow status```, your library (the sender) does not raise an error. This is a dump :

```
  slcan0  18DA2211   [8]  10 85 84 A6 73 65 6E 73
  slcan0  18DA1122   [3]  32 00 0A <- 3 == flow frame, 2 == OVERFLOW
```

And the python outputs just:

```
Payload transmission done.
```
Relevant paragraph in the 2004 ISO is 6.5.3.3. Let me know what you think, I might misinterpret the ISO as well (but the fact is that I send a message and even though it didn't pass through, I am not being informed abut that). Thanks.

pylessard, 2019-11-26:
Hi,
Thank you for the high quality bug report. I haven't reproduced your issue, but looking at the unit test, I see nothing that test that specific error, which should mean I forgot to do it (I normally unit test every feature.) 

I agree that this error should be reported. I will review soon and probably fix within a week or 2.

Regards

pylessard, 2019-11-26:
Confirmed, this is not reported.

https://github.com/pylessard/python-can-isotp/blob/44aa04aed84ed88e5b4de5dc1095960cc7b484cb/isotp/protocol.py#L526-L529

pylessard, 2019-11-26:
I just pushed a fix. Can you try?

iwasz, 2019-11-27:
It works! Thanks!

pylessard, 2019-11-27:
Good, I will publish the update tonight.
Thank you

commit 636e3c1b (2019-11-26): Now report reception of Overflow FlowControl PDU. Fixes #17
closed 2019-11-27 by pylessard