pylessard/python-can-isotp #18: How to use and The result?

Qiucy-Shi, 2019-11-28:
I am confused with the use of such package and the result these codes will provide.

pylessard, 2019-11-28:
Hi,
In order to get a good response, you need to ask a good question.
I do not know what you do not understand. Is it

- You don't know how to use Python?
- You don't know how to install a package with pip?
- You don't know what ISO-15765 protocol is and what it does?
- You don't understand the examples in the documentation?

regards

NullPointerException00, 2019-11-29:
Maybe my question needs to be more specific. You must know the PCAN-Basic API, it provides the python sample to use, I just need to run the .py file. I would like to know whether your package has the UI interface and then I just need to run a certain .py file to make the can bus work. If still confused about my question, freely to ask me. Thx

pylessard, 2019-11-29:
This package does not have a UI. It simply implements the ISO-15765-2 protocol.
There are examples in the documentation that you can run.

https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can

The problem you ask is not specific enough. "Make the bus work" is not a clear request. If you use peak interface, you can use this module with python-can module or, if you work on Linux (which I doubt), you can use it with [this kernel module](https://github.com/hartkopp/can-isotp).

If you want a UI, there a plenty of tools that can interact with your hardware out there. One that is free is [BusMaster.](https://rbei-etas.github.io/busmaster/).

Unless you have a specific question, I cannot help much.

NullPointerException00, 2019-12-01:
I understood. You have helped a lot, thx.

closed 2019-12-01 by NullPointerException00

NullPointerException00, 2019-12-23:
When I run the Basic transmission with python-can in https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can
An error occurred:
**Could not import vxlapi: Could not find module 'vxlapi'. Try using the full path with constructor syntax.
Traceback (most recent call last):
  File "C:/Work/python-can-isotp-master/isotp/test.py", line 10, in <module>
    bus = VectorBus(channel=0, bitrate=500000)
  File "C:\python3.7\lib\site-packages\can\interfaces\vector\canlib.py", line 81, in __init__
    raise ImportError("The Vector API has not been loaded")
ImportError: The Vector API has not been loaded**

I am new to python and I have no idea with such a problem, could you give me a help?

NullPointerException00, 2019-12-23:
I now use the Peak's device, the PCAN usb, https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can  
I would like to use your great isotp project to send and get big size msg. Could I use the basic transmission example directly, since I am new to these things, I really do not how to connect these tools and send, receive info.

pylessard, 2019-12-25:
You can use the example almost as is. Use a ``can.interfaces.pcan.PcanBus`` instead of a ``can.interfaces.vector.VectorBus``.    What you need to understand is that python-can is a different project that take care of interacting with your hardware. This module (python-can-isotp) is independent of the hardware but you need to give it a mean of sending  and receiving messages. 

When used with python-can, this mean is the ``Bus`` object.  First, try to send and receive 8 bytes message only with python-can. If you get this to work, you'll have a good start. You can always refer to python-can'S doc and ask question there if you have trouble with their module 

https://python-can.readthedocs.io/en/master/index.html

Regards