pylessard/python-can-isotp #19: Access underlying python-can bus

Oliverleo123, 2019-12-10:
I am trying to use can-isotp with udsoncan.
The isotp layer should only need those messages with the specified receive message id.
In my use case I must be able to receive all other messages and process them independent of isotp and without interfering uds/isotp communication.

I couldnt find any documentation about this behaviour.
Is this possible with the current implementation of isotp?

I tried using a second thread which calls the receive method of the can bus. But then I will get a timeout exception when I use uds services.

I am posting in the isotp project because I think this would be a feature for the isotp layer.

pylessard, 2019-12-10:
Yes this can be done, but will be handled at the link layer level.
If you are using python can, you can put a filter on your bus.

Maybe you could share some of your code so that I could understand the issue correctly.

pylessard, 2019-12-14:
Should this issue be closed?

Oliverleo123, 2019-12-17:
I will post my example code tomorrow

Oliverleo123, 2019-12-18:
```python

class CANReceiver(threading.Thread):

    def run(self):
        for i in range(1000):
            message = bus.recv(timeout=0.1)
            if isinstance(message, can.Message):
                print(message)
                x = 0
            else:
                print("timeout")

can.rc['interface'] = 'pcan'
can.rc['channel'] = 'PCAN_USBBUS1'
can.rc['state'] = can.bus.BusState.ACTIVE
can.rc['bitrate'] = 500000

try:
    bus = can.ThreadSafeBus()
except can.interfaces.pcan.pcan.PcanError as e:
    print(e)
    sys.exit(-1)

isotp_params = {
   'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
   'squash_stmin_requirement' : False     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
}

tp_addr = isotp.Address(addressing_mode=AddressingMode.Mixed_11bits, txid=0x10A, rxid=0x10B, address_extension=0)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(isotp_layer=stack)

config = dict(udsoncan.configs.default_client_config)
config['p2_timeout'] = 1
config['security_algo'] = RC4_algo
config['security_algo_params'] = security_keys.keys['bmz_end_of_line']

thread = CANReceiver()
thread.start()

with Client(conn, config=config) as client:
        client.change_session(0x11)

```

the line `client.change_session(0x11)` always has a timeout when Im using the the CANRecevier Thread.

What I want to achieve is to use the bus for both isotp (uds) and normal messages

Oliverleo123, 2019-12-18:
```python

class CANReceiver(threading.Thread):

    def run(self):
        for i in range(1000):
            message = bus.recv(timeout=0.1)
            if isinstance(message, can.Message):
                print(message)
                x = 0
            else:
                print("timeout")

can.rc['interface'] = 'pcan'
can.rc['channel'] = 'PCAN_USBBUS1'
can.rc['state'] = can.bus.BusState.ACTIVE
can.rc['bitrate'] = 500000

try:
    bus = can.ThreadSafeBus()
except can.interfaces.pcan.pcan.PcanError as e:
    print(e)
    sys.exit(-1)

isotp_params = {
   'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'll_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00. None means no padding
   'squash_stmin_requirement' : False     # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
}

tp_addr = isotp.Address(addressing_mode=AddressingMode.Mixed_11bits, txid=0x10A, rxid=0x10B, address_extension=0)
stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
conn = PythonIsoTpConnection(isotp_layer=stack)

config = dict(udsoncan.configs.default_client_config)
config['p2_timeout'] = 1

thread = CANReceiver()
thread.start()

with Client(conn, config=config) as client:
        client.change_session(0x11)

```

the line `client.change_session(0x11)` always has a timeout when Im using the the CANRecevier Thread.

What I want to achieve is to use the bus for both isotp (uds) and normal messages

Oliverleo123, 2019-12-18:
After taking a look into the source code, I think the problem could be solved by creating a new subclass of bus (a virtual bus), which will receive messages from a queue.
isotp.CanStack would be instantiated with this virtual bus.

The "real" can bus would then poll and receive all messages and put those with the IDs needed for isotp into the queue of the virtual bus. The isotp.process method would receive all mesages from that queue.

Also, transmit messages would need to be routed from the virtual bus of the isotp.CanStack to the real bus.

Doesnt seem to be the cleanest solution, but could work for my intended use. Do you have another suggestion?

pylessard, 2019-12-19:
Actually, you are not the first one to ask such question. Right now, the module has everything needed to make this work, I will probably need to improve the doc to explain this better.

There is 2 options
1 - Use 2 Bus object and put filters on them
2 - make a message routin mechanism. This can be done in many ways, subclassing the Bus object is one. Another one would be to use cuatom rxfn and txfn with the TransportLayer object.

Note that you can use the address is_for_me method to check if a message is to be received by the isoyp layer. I could probably add something that check transmission as well.

Does that helps?

Oliverleo123, 2019-12-19:
Option 1 would be my preferred way, but I dont think I can initialize the same bus 2 times, at least not with the PCAN interface.
But I think using a routig mechanism should be no problem.

Thanks for your help :)

closed 2019-12-20 by pylessard