pylessard/python-can-isotp #20: Tx Frames sent at STmin = 0 while setting Tx STmin to a value > 0

wanam, 2019-12-23:
I'm trying to set Tx STmin to a value between 1ms and 10ms, but with no success.

``` python
_connection = IsoTPSocketConnection(self._can_channel, rxid=self._fota_uds_rxid, txid=self._fota_uds_txid, addressing_mode=addressing_mode)
_connection.tpsock.set_opts(optflag=optflag, txpad=0, frame_txtime=self._txdelay, tx_stmin=5)
```

Current behavior:
1- If we set Tx STmin to `None`, we respect the server Rx STmin value which is 10ms in my case, which is working fine.
2- But when I set STmin > 0ms, we ignore the server STmin value, but I see in the candump that the frames are sent at 0ms for what ever non null value I set for STmin.

pylessard, 2019-12-24:
Hi, 
I confirm that I have been able to reproduce the problem. I will investigate and feedback to you as soon as possible.

Regards

pylessard, 2019-12-25:
Hi,
After deeper review, the module is working fine. The thing is that the ``tx_stmin`` value is in nano seconds. You seems to have assumed that this value is set in millisecs (and that's what I did also). You should put 5000000 to get 5 millisecs.

See the below screenshot taken directly from the kernel module source.  I will update the to avoid the caveat.

[image]

wanam, 2019-12-26:
Understood, thank you for your support!

wanam, 2019-12-26:
Hi @pylessard ,

I made a test with the python can-isotp module, and it seems stmin as described in [this example](https://udsoncan.readthedocs.io/en/latest/udsoncan/examples.html#using-uds-over-python-can) needs to be 0-127ms.
https://github.com/pylessard/python-can-isotp/blob/636e3c1b0171fcee2e7ad9c912eda8c8942faa73/isotp/protocol.py#L243-L244

But it does not seem to work, I get the same behavior as I described for the kernel module, ie with `None` or `0` value it is Ok, but with values > 0 I always get the Tx frames sent at ~0ms, even with values at nano scale!

It is not a big deal for me since I'm using the kernel module, but I wanted to report it for people that may need to use the python module.

pylessard, 2019-12-26:
Hi @wanam,
The configuration you have pointed out is used for the TransportLayer objectg (python implementation in the user space). When using the an IsoTP socket (kernel module), you should refer to this : 
https://can-isotp.readthedocs.io/en/latest/isotp/socket.html#socket.set_opts

I've updated the documentation to specify the timing units.

Now, I am a little confused as why is doesn't work for you. I'd like to find why.  ``tx_stmin`` must be set on the socket that sends data, is that what you do?  

And last thing, when specifying the address for the socket, it would be better to pass an Address object instead of txid, rxid. The way you do is working only for backward compatibility, but passing an Address object is preferred. You may have trouble if you change addressing mode.

Will wait for your input.

wanam, 2019-12-26:
Hi @pylessard ,

Actually the issue I am talking is about the python implementation in the user space, for the kernel module it is working fine with nano values as you pointed out.

I will double check the configuration I'm using for the python implementation.

pylessard, 2019-12-26:
Hi @wanam,
The user space implementation does not have a parameter that reproduce the tx_stmin parameter in the ketnel module.

The stmin parameter you referred to is the value given by the receiver in a flow conyrol message.

There is the squash_stmin_requirement parameter that will force the stack to ignore the receiver stmin value. This parameter is a boolean and will make the stack transmit as fast as possible when True. It would be hard to override the timing with a precise numerical value in the user space. I could implement the feature, but it would not work in most case. For example, on Windows, it would be either 0ms or 16ms.

Hope that helps

wanam, 2019-12-26:
Oki I thought it is supprted to customize the timing with the user space module as well, no biggie, thank you again for your time.

closed 2019-12-26 by wanam