pylessard/python-can-isotp #21: Getting timeout while 4KB packet over normal CAN

wanam, 2020-01-13:
I am using this lib with the can-isotp kernel module, and I try to start a data transfer 0x36 request with max data of 4KB over normal CAN (or with CAN-FD while setting tx_dl to 8 bytes) and STmin = 10.

During my tests, sending this packet of multi-frame takes more than 5 seconds to complete.

So I had to increase the timeout (0.1) set bellow to 10 seconds to avoid a socket timeout:
https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/connections.py#L260

According to UDS standard, there are no limitations for the sending duration, and according to a comment you put [here](https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/connections.py#L155), this timeout was meant for receive only, but I don't understand how it you should be able to do that, the socket's timeout will be triggered for the sending as well if it passes the 100ms.

But increasing this timeout is not enough, because I am getting a P2 timeout, because somehow the socket->send reports ([here](https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/client.py#L1469)) completed within 2 or 3 seconds, and it returns the full packet size 4KB, while the sending still continues at kernel module level! this causes the timer to start for the receive and [raises](https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/client.py#L1496) a timeout while the ECU is still waiting for the remaining frames of the packet.

I debugged the [can-istop kernel module](https://github.com/hartkopp/can-isotp) to see if it is reporting a complete status before it ends sending all the consecutive frames of the 4K packet, but it seems the timing seems correct at this module level ( > 5 seconds):

[178158.310957] can-isotp: isotp_sendmsg: starting txtimer for fc
[178159.310929] can-isotp: isotp_tx_timer_tsklet: we did not get FC frame in time.
[178159.310949] can-isotp: isotp_sendmsg: starting txtimer for fc
[178159.312847] can-isotp: isotp_rcv_fc: FC frame: FS 0, BS 0, STmin 0x0A, tx_gap 10000000
[178159.312849] can-isotp: isotp_rcv_fc: starting txtimer for sending
[178159.322853] can-isotp: isotp_tx_timer_tsklet: next pdu to send.
[178159.332868] can-isotp: isotp_tx_timer_tsklet: next pdu to send.
......
[178165.164485] can-isotp: isotp_tx_timer_tsklet: next pdu to send.
[178165.174498] can-isotp: isotp_tx_timer_tsklet: next pdu to send.
[178165.174505] can-isotp: isotp_tx_timer_tsklet: we are done

@pylessard @hartkopp could you help? Being new to Python and have a very limited knowledge on C++, I'm maybe missing something about how the socket works on both layers.

pylessard, 2020-01-13:
Hi @Wanam,
The timeout value you are changing is the timeout of the reception thread, it is not related to UDS P2 or P2* timeout. This is just the rate at which the thread polls the socket for new data. You shouldn't have to changes this value.

In your log, I doi not see the timeout error, but you mentioned that you got a P2 timeout which is defined by UDS, not Iso-TP. The UDS timeout is set here : https://github.com/pylessard/python-udsoncan/blob/master/udsoncan/client.py#L1488

Right now, the timeout to get a response from the server starts at the beginning of the transmission. Maybe this isn't correct, but I don't have the latest UDS standard at hand and timeout are defined in it.  I have the 2006 version for now.  I will double-check that as soon as I can.

You can overcome your limitation by modifying the UDS timeouts (there is 3 of them), refer to the documentation here : https://udsoncan.readthedocs.io/en/latest/udsoncan/client.html#request_timeout

Let me know if that helps.

pylessard, 2020-01-13:
Actually, your issue is possibly a duplicate of this issue : https://github.com/pylessard/python-udsoncan/issues/42

Only reason why I do not work on it is the fact that I don't have the standard at hand yet.

wanam, 2020-01-13:
> pylessard/python-udsoncan#42

Hi Pylessard,

I think this is a different issue, in this case we have a single ack frame
on the response, and the main issue in my case is that the P2 timer starts
before the send is actually complete.

The timeout value I pointed before is applied also to the socket not only
the receive thread:
https://github.com/pylessard/python-can-isotp/blob/master/isotp/tpsock/__init__.py#L76

I didn't include the socket send timeout error in my logs, I will add the
logs later, but it's happening here:

https://github.com/pylessard/python-can-isotp/blob/master/isotp/tpsock/__init__.py#L81