pylessard/python-can-isotp #22: Linux kernel 5.4.x displays "OSError: [Errno 22] Invalid argument" when _socket.bind() called

rslindee, 2020-01-14:
After recently updating my Linux kernel to 5.4.x, it seems that there is some sort of interface change to binding sockets causing breakage. As a result I get the following traceback:

```
Traceback (most recent call last):
  File "./my-app", line 22, in <module>
    my_session.start()
  File "/home/rslindee/work/my_session.py", line 112, in start
    self.client.open()
  File "/home/rslindee/work/python-udsoncan/udsoncan/client.py", line 58, in open
    self.conn.open()
  File "/home/rslindee/work/python-udsoncan/udsoncan/connections.py", line 260, in open
    self.tpsock.bind(self.interface, rxid=self.rxid, txid=self.txid)
  File "/home/rslindee/work/python-can-isotp/isotp/tpsock/__init__.py", line 165, in bind
    self._socket.bind((interface, rxid, txid))
OSError: [Errno 22] Invalid argument
```

If I revert my kernel to 5.3.x, everything works fine (this is an existing application that utilized your python modules with great success up until now). Note: I am able to successfully build my isotp kernel module and run a tool like isotpdump just fine with kernel 5.4.x, it is primarily my python script leveraging python-can-isotp that I'm having trouble with.

Some more system info:

OS: Fedora 31
Kernel: 5.4.8
Python: 3.7.6

Please let me know if there is any additional info I could provide that would be useful!

pylessard, 2020-01-15:
Hi, thank you for this bug report. This seems a little tricky to handle as it looks like a mismatch between Python and the kernel module.

I think the problem arise from the fact that the ``sockaddr_can`` in the kernel changed size because of [this commit](https://github.com/torvalds/linux/commit/f5223e9eee651e005c0f6d6d078909087601b7e9#diff-0e0481b441394a0e77d56275d10c13f7)

Now the Python socket module creates the address ([Here](https://github.com/python/cpython/blob/master/Modules/socketmodule.c#L2208)) based on the linux header it had when compiled, in your case, most likely without J1939 support.

Finally, the ISO-TP kernel validates the address when ``bind`` is called by checking it size given by the user against the ``sockaddr_can`` object size. In your case, if you compile with linux 5.5, the size of the structure should be bigger than what you Python knows because it includes J1939.

Now, how to fix? I am not sure what's the best approach. If you can get a Python compiled for Linux 5.5, it should work. You can try compiling python yourself and try.
I think the best approach would be to ask @hartkopp to update the kernel module to support this case. I will open an issue on his repo.

rslindee, 2020-01-15:
Not a problem - I'm a big fan of your efforts with this and appreciate the quick response.

One thing I just want to note is that this was found with (and built with the headers of) Linux kernel 5.4, not 5.5. Your explanation definitely sounds plausible and directed in the right place, but from what I'm reading, it seems like the J1939 addition was specifically introduced in the 5.5 kernel, not 5.4. Full disclaimer: I'm a bit out of my element here, so by all means feel free to disregard my suggestion if I'm missing something.

I'll keep an eye on the can-isotp you opened, as well. Thank you again and let me know if there is anything else I can do to assist.

pylessard, 2020-01-15:
My bad, I meant 5.4, not 5.5
I found this changelog for 5.4 that state the addition of J1939 : https://kernelnewbies.org/Linux_5.4

I may miss something though, I rarely have to look at the Linux source :)

rslindee, 2020-01-15:
No worries and thanks for confirming!

hartkopp, 2020-01-15:
@pylessard : Thanks for the hint! Really missed to update can-isotp while acking the mainline patches m(
Added 5.4 specific handling on master branch in https://github.com/hartkopp/can-isotp/commit/fbed98a0b5de1bbb29479516cf9b4aeb8b52ec55
@rslindee : Can you check, if it fixes the problem for you? Tnx!

hartkopp, 2020-01-15:
Just got behind the real requirement:
https://github.com/hartkopp/can-isotp/commit/ced84cab120a1ddde2bd4561a18514707c8a6184

rslindee, 2020-01-15:
Tested by updating can-isotp to the latest commit, rebuilt and things are now working great. No change was needed on the python side of things.

Thank you both for the quick turnaround - you're both doing great work!

closed 2020-01-15 by rslindee

pylessard, 2020-01-15:
Thank you guys!

hartkopp, 2020-01-16:
Yes. Thanks to both of you!