pylessard/python-can-isotp #23: Non python canbus usage

Jens2F4711, 2020-03-16:
I want to use python-can-isotp (for python uds) with continuing to use the python canbus for other non iso-tp messages. 

I have tested one of the solutions for this mentioned in #19 .

I have created a virtual canbus class inherited from BusABC which re-implements the 'recv' and 'send' methods. The virtual canbus class itself is taking a notifier and a canbus object. This bus is passed to 'isotp.CanStack'.

I can see on the canbus that the rested Id is correctly sent and that the response id is received (via the listener object). 
The isotp is calling / pooling the 'recv' of the virtual canbus but it seems not to process the data because there is a timeout error for the isotp response.

If a pass a real python can bus object isotp and uds is works correctly.

pylessard, 2020-03-19:
Could you share some code?
Also, I would suggest to use filter on the bus object or doing a dispatching mechanism with queues instead.

Jens2F4711, 2020-03-19:
I think the most important class of my code is the `virtual_canbus` module / `VirtualCanbus` class which is passed to isotp.CanStack().  
What do you mean by filters? If a filter the can messages of the bus object by the response ID and call some function: How should I passed / dispatch them to isoTp class? 
```
import can
from can import BusABC
from can import Listener
from can import Notifier
from waiting import wait
import threading
from can.message import Message
from typing import cast, Iterator, List, Optional, Sequence, Tuple, Union

def override(f):
    return f

class MyListener(Listener):
    def on_message_received(self, msg):
        VirtualCanbus.add_to_stack(msg=msg)

class VirtualCanbus(BusABC):
    def __init__(self, canbus_and_notifier):
        self.__canbus_and_notifier = canbus_and_notifier
        self.__my_listener = MyListener()
        self.__canbus_and_notifier[1].add_listener(self.__my_listener)

    def __del__(self):
        print("VirtualCanbus.del()")
        self.close()

    def close(self):
        self.__my_listener.stop()
        del self.__my_listener
        print("VirtualCanbus.close(self):")

    @classmethod
    def add_to_stack(self, msg):
        self.__msg = msg
        self.__cv.acquire()
        self.__cv.notify()
        self.__cv.release()

    @override
    def recv(self, timeout: Optional[float] = None) -> Optional[Message]:
        # function is polled by iso_tp (resulting high cpu load)
        self.__cv.acquire()
        self.__cv.wait(timeout=timeout)

        if self.__msg is not None:
            if self.__msg.arbitration_id == 0x10B:
                print("found")  # Message is found here
                print(self.__msg)

        return self.__msg

    @override
    def send(self, msg: Message, timeout: Optional[float] = None):
        print("send")
        print(msg)
        self.__canbus_and_notifier[0].send(msg, timeout)

    __msg = None
    __cv = threading.Condition()
```

uds_function.py
```
from uds_services import UdsServices

class UdsFunction():
    def __init__(self, canbus_and_notifier):
        self.__canbus_and_notifier = canbus_and_notifier

    def get_uds_interface(self, request_id, response_id, config):
        self.__uds_services = UdsServices(self.__canbus_and_notifier,
                                          request_id,
                                          response_id,
                                          config)
        return self.__uds_services

    def close(self):
        self.__uds_services.close()
        del self.__uds_services
```

uds_services
```
from virtual_canbus import VirtualCanbus
import isotp
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.connections import IsoTPConnection
from udsoncan.client import Client
from udsoncan.exceptions import *
from udsoncan.services import *
from functools import partial
from can.message import Message

class UdsServices(Client):
    def __init__(self, canbus_and_notifier, request_id, response_id, config):
        self.__setup_iso_tp(canbus_and_notifier, request_id, response_id)
        super().__init__(self.__conn, config=config, request_timeout=0.5)

    def __setup_iso_tp(self, canbus_and_notifier, request_id, response_id):
        print("---------------------")
        print(type(canbus_and_notifier))
        self.__special_bus\
            = VirtualCanbus(canbus_and_notifier=canbus_and_notifier)
        print("type(special_bus)")
        print(type(self.__special_bus))
        tp_addr = isotp.Address(
            addressing_mode=isotp.AddressingMode.Mixed_11bits,
            txid=request_id,
            rxid=response_id,
            address_extension=0)

        stack = isotp.CanStack(self.__special_bus, address=tp_addr)
        self.__conn = PythonIsoTpConnection(stack)

    def close(self):
        self.__special_bus.close()
        del self.__conn
```

```
#!/usr/bin/env python

import udsoncan
from typing import Final
from can import Listener
from can import AsyncBufferedReader
import time
import can
from can import Notifier

from uds_function import UdsFunction

class MyListener(AsyncBufferedReader):
    def on_message_received(self, msg):
        if self.__i % 100 == 0:
            print(self.__i, ":", msg)
        self.__i += 1

    __i = 0

class DataPrinter(udsoncan.DidCodec):
    def encode(self, val):
        return val

    def decode(self, payload):
        return payload

    def __len__(self):
        return 1

config = dict(udsoncan.configs.default_client_config)
DATA_IDENTIFIER_CELL_VOLTAGE: Final = 0x7010
DATA_IDENTIFIER_NUMBER_OF_SLAVES: Final = 0x4005  # 0x6405
DATA_IDENTIFIER_NUMBER_OF_MASTER: Final = 0x4006
DATA_IDENTIFIER_BATTERY_LIM_VOLTAGE_MIN: Final = 0x4072
data_indetifier = DATA_IDENTIFIER_NUMBER_OF_SLAVES

config['data_identifiers'] = {
   DATA_IDENTIFIER_CELL_VOLTAGE: DataPrinter(),
   DATA_IDENTIFIER_NUMBER_OF_SLAVES: DataPrinter(),
   DATA_IDENTIFIER_NUMBER_OF_MASTER: DataPrinter(),
   DATA_IDENTIFIER_BATTERY_LIM_VOLTAGE_MIN: DataPrinter()
   }

can_bus = can.ThreadSafeBus(bustype='socketcan',
                            channel="can0",
                            bitrate=500000)

my_listener = MyListener()
canbus_and_notifier = (can_bus, Notifier(can_bus, [my_listener]))
canbus_and_notifier[1].add_listener(my_listener)

uds_function = UdsFunction(canbus_and_notifier)
client = uds_function.get_uds_interface(0x10A, 0x10B, config)

client.open()
try:
    response = client.read_data_by_identifier(data_indetifier)
    print(response.service_data.values[data_indetifier])
except udsoncan.TimeoutException:
    print("TimeoutException")

client.close()
```

pylessard, 2020-03-19:
See https://python-can.readthedocs.io/en/master/bus.html#can.BusABC.set_filters
Basically, you instruct your bus object to consume only message with specific ID so the other messages are not consumed then lost.

Jens2F4711, 2020-03-24:
Basically I do not understand how a filter can help: I have one python bus object that is receiving from physical can bus. I want to able to receive all messages (maybe also including the iso tp ones) outside the iso tp class. If I put filters on it consuming only the iso tp response id before passing the python bus object to iso tp class I would not get all the other messages outside the iso tp class. Filtering for all but the response id would disable the iso tp functions.

I have tried another approach: I tried using 2 additional virtual can buses. The first one is passed to the iso tp class. By using listeners I am passing messages from the real python can bus to the second virtual bus and vice versa. This is working - the iso tp is working. But:
1.  It seems that here a listener on the real python canbus is also consuming messages so that the cannot reached the iso tp class.
2. Also without that listener or without passing to iso tp (and before the virtual buses) the python can seems not receives all messages which are incoming (checked with a second can interface and via candump in a terminal).

But both seems / maybe not related to iso tp.

Jens2F4711, 2020-03-24:
Finally the approach with 2 additional virtual can buses is working now.

Jens2F4711, 2020-04-07:
Maybe this code can help somebody else:

```
import can
from can import AsyncBufferedReader, Notifier

class PhysicalBusListener(AsyncBufferedReader):
    def __init__(self, virtual_to_physical_bus, response_id):
        self.__virtual_to_physical_bus = virtual_to_physical_bus
        self.__response_id = response_id

    def on_message_received(self, msg):
        if msg.arbitration_id == self.__response_id:
            self.__virtual_to_physical_bus.send(msg)

class IsoTpVirtualBusListener(AsyncBufferedReader):
    def __init__(self, physical_bus, request_id):
        self.__physical_bus = physical_bus
        self.__request_id = request_id
        self.__last_msg = None

    def on_message_received(self, msg):
        if msg.arbitration_id == self.__request_id\
                and msg != self.__last_msg:
            self.__physical_bus.send(msg)
            self.__last_msg = msg

class CanbusDispatching():
    def __init__(self,
                 canbus_and_notifier,
                 request_id,
                 response_id,
                 canbus_name):
        virtual_to_physical_bus = can.interface.Bus(canbus_name,
                                                    bustype='virtual')
        physical_bus_listener = PhysicalBusListener(virtual_to_physical_bus,
                                                    response_id)
        canbus_and_notifier[1].add_listener(physical_bus_listener)

        iso_tp_virtual_bus_listener\
            = IsoTpVirtualBusListener(canbus_and_notifier[0],
                                      request_id)
        notifier_on_virtual_to_physical_bus\
            = Notifier(virtual_to_physical_bus,
                       [iso_tp_virtual_bus_listener])
        notifier_on_virtual_to_physical_bus.add_listener(
            iso_tp_virtual_bus_listener)
```

```
import isotp
from udsoncan.connections import PythonIsoTpConnection, IsoTPConnection
from .canbus_dispatching import CanbusDispatching

CANBUS_NAME: Final = "isotp"
virtual_to_isotp_bus\
        = can.interface.Bus(CANBUS_NAME, bustype='virtual')
canbus_dispatching = CanbusDispatching(canbus_and_notifier,
                                       request_id,
                                       response_id,
                                       CANBUS_NAME)
tp_addr = isotp.Address(
        addressing_mode=isotp.AddressingMode.Mixed_11bits,
        txid=request_id,
        rxid=response_id,
        address_extension=address_extension)
stack = isotp.CanStack(virtual_to_isotp_bus, address=tp_addr)
connection = PythonIsoTpConnection(stack)
```

closed 2020-04-07 by Jens2F4711