pylessard/python-can-isotp #25: Receiving from multiple addresses in NormalFixed addressing

lrabe, 2020-04-03:
My goal is to use the OBD-2 protocol on top of can-isotp. 
In OBD-2, the target/source addresses are encoded in the arbitration ID, therefore I use AddressingMode.NormalFixed_29bits

I want to receive both messages that are "broadcasted", i.e. messages with an arbitration id of 0x18DB33F1, as well as messages that are directed to one specific receiver, e.g. 0x18DA01FF.

I tried using two separate CanStacks in order to receive both types of messages, like so:

```
addr1 = isotp.Address(isotp.AddressingMode.NormalFixed_29bits, target_address=0xF1, source_address=0x33)
stack1 = isotp.CanStack(self.bus, address=addr1, error_handler=self.my_error_handler)

addr2 = isotp.Address(isotp.AddressingMode.NormalFixed_29bits, target_address=0xF1, source_address=0x01)
stack2 = isotp.CanStack(self.bus, address=addr2, error_handler=self.my_error_handler)
```
For each stack I have a separate thread that calls its `process()` method.  
With this, I do not receive any messages. If I only use the first two lines, I am receiving the broadcast messages correctly.

Is there a better way to do that?

pylessard, 2020-04-14:
Hi, sorry for the late response, I missed that issue.
The problem is that you use the same bus for the two stack, so one stack consume the other ones messages.

Option 1 : you use 2 different bus, you can add filters on these busses so that they read the same hardware bus but don't read the message not meant for them.

Option 2 : Make a dispatching mechanism. Make a thread that read the message from your bus and then input these message in the TransportLayer object. If you do so, you can't use directly the CanStack object. You will need to use the TransportLayer object and define custom ``rxfn`` and ``txfn``

I would go for option 2.  If you have 2 stacks, you don't need necessarily 2 threads, tou can have one thread that call the process method on both stacks. 

Can you help me understand in what context you need to receive the 2 types of messages?  I am asking because there is pull request #16 that could handle this within the same stack, but I haven't merged it because I couldn't find any statement in ISO-15765 justifying this change. If it is industry standard to accept both functional and physical message, I could reconsider.

lrabe, 2020-04-14:
Thank you for the response! I'll try implementing Option 2. 

My context is that I want to simulate an ECU that communicates with a diagnostic OBD2 device.  

The diagnostic device will send broadcast messages (functional addressing) when it does not know which ECU should reply. The arbitration ID is then 0x18DB33F1.    
The diagnostic device will send directed messages (phyical addressing) to a specified ECU when it knows which ECU should reply. The arbitration ID is then (for example) 0x18DA01F1. 

The simulated ECU should be able to receive both of these messages. 

Pull request #16 appears like it only handles this case for sending both types of message, not receiving them - is that correct or did I miss something?

lrabe, 2020-04-16:
@pylessard I've used the following workaround to be able to receive both types of messages:

```
class CustomAddress(isotp.Address):
   def _is_for_me_normalfixed(self, msg):
      BROADCAST_ADDRESS = 0x33

      if self.is_29bits == msg.is_extended_id:
         is_from_correct_target = msg.arbitration_id & 0xFF == self.target_address

         if ((msg.arbitration_id >> 16) & 0xFF) == 0xDA: # physical addressing
            is_for_my_address = (msg.arbitration_id & 0xFF00) >> 8 == self.source_address
            return is_for_my_address and is_from_correct_target
         elif ((msg.arbitration_id >> 16) & 0xFF) == 0xDB: # functional addressing
            is_broadcast = (msg.arbitration_id & 0xFF00) >> 8 == BROADCAST_ADDRESS
            return is_broadcast and is_from_correct_target
      return False

addr = CustomAddress(addressing_mode=AddressingMode.NormalFixed_29bits, target_address=0xF1, source_address=0x01)
stack = isotp.CanStack(self.bus, address=addr, error_handler=self.my_error_handler)
```

I just override the `_is_for_me_normalfixed` function so it also accepts messages that are sent to the functional address 0x33.

pylessard, 2020-04-16:
Carefull with that. You may get PDU from different ISOTP frames mixed up.
The idea of having 2 stack is having two independent receive buffer.

lrabe, 2020-04-16:
Good point. I think in my case it should be fine since there's only two devices on the bus and all communication happens sequentially. 

Still, it's just a workaround, not a proper solution.

pylessard, 2020-06-05:
I think this issue can be closed

closed 2020-06-05 by pylessard