pylessard/python-can-isotp #28: Send Normal_11bits frame receive Normal_29bits frame

DanielKupiniak, 2020-05-14:
Hi, is it possible to mix can id types for one stack? I want to send a standard frame but receive extended.

pylessard, 2020-05-14:
Hi, it is not possible to do so right now. 

A quick workaround would be to convert the 29 bits ID to an arbitrary 11bits ID that the stack will accept (or the other way around).

I will look a the feasability to implement this during the weekend.

DanielKupiniak, 2020-05-14:
Thanks for the tip, I will try to use it

DanielKupiniak, 2020-05-15:
What do you think to create a class:
```
class Address.CanID(object):
     self.id
     self.addressingMode or self.is_extended
```
I know it will break interface, but also will limit the number of arguments to Address cls
What I see from the code and thanks to its modularity changes needs to be applied only in Address cls
<Good Job>

pylessard, 2020-06-08:
Hi, there seems to be no gracveful solution to this that won't break the interfaces.
I will come up with something, but it may take some time. Sorry if there's a delay, I don't have much time now.

pylessard, 2023-10-23:
I'm looking into this now. Will be possible in v2.0

pylessard, 2024-01-03:
This is implemented in PR #107

pylessard, 2024-01-03:
Fixed by #107. Will be part of v2.x
Release will be done in January 2024

closed 2024-01-03 by pylessard