pylessard/python-can-isotp #29: can.Message.__init__, is_extended_id vs extended_id

ericevenchick, 2020-06-05:
I'm having trouble using python-can-isotp with python-can. `tx_canbus` attempts to create a new `can.Message` using the `extended_id` keyword argument, but this has been changed to `is_extended_id`.

The issue is at https://github.com/pylessard/python-can-isotp/blob/d24462ea1faf3e39904879ea3d30484d3891aed8/isotp/protocol.py#L871

It seems like a simple one line fix, but I'm not sure if I using the wrong version or doing something else to cause this error.

pylessard, 2020-06-05:
Apparently, Python-can has renamed the parameter. 
I will update the library tonight

Thank you!

[image]

pylessard, 2020-06-05:
Can you try the fix I just pushed?

ericevenchick, 2020-06-06:
Looks like it works now, thanks for the quick fix!

pylessard, 2020-06-08:
Fix published in v1.6

commit d8226824 (2020-06-05): Fixed incompatibility of python-can 4+ with regards of Message.is_extended_id naming. Fixes #29
closed 2020-06-08 by pylessard