pylessard/python-can-isotp #33: Functional Message Support for Server

CanBusHack, 2020-09-16:
I need to receive a message as a functional request (0x7DF) however I need to response on a physical address (0x7E8) and also listen for Flow Control Frames on (0x7E0).  

I would like to my isotp.Address(rxid=[0x7DF, 0x7E0]) so that I can catch any of these messages.  I'm working on updating this now but am not sure if I'll have success.

pylessard, 2020-09-17:
Hi, in order to receive functional messages and physical messages, I suggest to have two different TransportLayer object. It will keep things clean and make sure both have their own receive buffer.

As for sending Flow Control with a different ID; this isn't specified by the standard, so it won't be supported natively. Still, you can work this around. I suggest you change the ID yourself in your transmit function ``txfn``.   You can read the CAN message that goes out of the TransportLayer and then change the ID based on the content.  You can use isotp.protocol.PDU class to parse the message.  Something like

    pdu = isotp.protocol.PDU(msg)
    if pdu.type == isotp.protocol.PDU.Type.FLOW_CONTROL:
        msg.id = 0x7E0

CanBusHack, 2020-09-17:
Thank you for your suggestion, I will attempt this.  However implementing a server (ECU Simulator) this solution isn't ideal.  As context switching shouldn't be required.

It is common for devices to support multiple RX Messages (Functional and Physical).  The first request being a Functional one, then the second request (flow control most likely, being a physical request).  I will work to implement the workaround.

CanBusHack, 2020-09-17:
After re-reading your workaround, I'm not sure how this might work? I'm trying to send a multiframe response (request I receive is a VIN Request from the client). It goes like this:

```
[Client]: 7DF 02 09 02 00 00 00 00 00
[Server]: 7E8 10 14 49 02 01 31 32 33
[Client]: 7E0 30 00 00 00 00 00 00 00
```

At this point no more messages are done, my code (server code) doesn't receive the 7E0 30 (Flow Control) because its filtered out.  If I add the `txfn` then I still don't see how I would receive this message to send the consecutive frames.  Am I missing something?

derek-will, 2021-05-02:
Hi @pylessard, I am running into a similar situation. I think I need to write a 1-to-n broadcast manager for handling responses from multiple ECUs that are solicited from a single functional request, but I hope to get feedback from you first.

For example:
7DF 02 01 00 <-- OBD Functional Request to read supported powertrain PIDs 01-20
7E8 06 41 00 XX XX XX XX <-- ECM response
7E9 06 41 00 XX XX XX XX <-- TCM response

In my test code, the CanStack that is processed first will take it's response and throw out the response from the other ECU when the process function is called on that CanStack. As the CanStacks are isolated from one another they don't know that they are throwing out messages that are for another CanStack. What I think is needed here is a Functional Request Manager that takes a bunch of CanStack objects as an argument and when processing the response queue passes the response to the correct CanStack for further processing.

There isn't a method already available for this? Is this the approach you would recommend for this project? Thanks for your time and awesome project.

pylessard, 2021-05-14:
Yes, I think you need a CanStack for each device to have a receive buffer per device. If they have different RX address, that shouldn't be an issue. If I understand well, your problem is mainly that one stack consume the message of the other and discard it because it is not meant for this stack.

There is at least 6 other issues that discuss this. The 2 solutions are either to have a bus object with filters on each stack or make your own dispatching mechanism as you proposed.

Hope that helps

tiger86023086, 2021-06-21:
I donnot know how to use the library to send FC ?

closed 2021-12-28 by pylessard