pylessard/python-can-isotp #36: can not enable data bitrate in CANFD mode

hphphpgithub, 2020-12-31:
in my test, after the CAN FD mode enable(TransportLayer['can_fd'] = true,), the FDF bit in CAN bus can be set, but the BRS bit can NOT be set. 
After my checking, there is no parameter **bitrate_switch** in the below can.Message() of protocol.py, but **bitrate_switch** is available parameter in the Message of python-can.

_class CanStack(TransportLayer):  
  def _tx_canbus_3plus(self, msg):
        self.bus.send(can.Message(arbitration_id=msg.arbitration_id, data = msg.data, is_extended_id=msg.is_extended_id, is_fd=msg.is_fd))
    def _tx_canbus_3minus(self, msg):
        self.bus.send(can.Message(arbitration_id=msg.arbitration_id, data = msg.data, extended_id=msg.is_extended_id, is_fd=msg.is_fd))_

is there any other method to enable BRS? or need update the source code of can-isotp

pylessard, 2020-12-31:
You are correct, you there is no option to set the BRS. Reason is: I did not knew about it.
I will add this option soon.

In the meantime, you can easily overcome that limitation by defining yourself the ``rxfn`` and ``txfn`` used by the TransportProtocol object.  The CanStack object is just a convenience that defines these callback for you, but you are not restricted to use it. Go with the TransportProtocol object directly and make your own ``rxfn`` and ``txfn``

Regards

pylessard, 2020-12-31:
Why don't you try this new branch?
https://github.com/pylessard/python-can-isotp/tree/Handle-CANFD-Bitrate-switch

Should do the trick. I will merge if you confirm it works as you intend.

Regards

hphphpgithub, 2021-01-04:
It works when config['bitrate_switch']=TRUE with new branch. pls go ahead. Thanks

pylessard, 2021-01-04:
Thank you!
Merged #38. Will make a release soon.

closed 2021-01-04 by pylessard