pylessard/python-can-isotp #37: Python script stucks at bind step.

kilianbr, 2021-01-04:
Hi, 
I've got some issues binding the sockets and using the socket example as described in the docs.
My code looks like the following:

```
import isotp

s = isotp.socket()
s2 = isotp.socket()
# Configuring the sockets.
s.set_fc_opts(stmin=5, bs=10)
#s.set_general_opts(...)
#s.set_ll_opts(...)

print("socket parameters set")

s.bind("vcan0", isotp.Address(rxid=0x123 ,txid=0x456))

print("bound first socket")
s2.bind("vcan0", isotp.Address(rxid=0x456, txid=0x123))
print("bound second socket")

s2.send(b"Hello, this is a long payload sent in small chunks of 8 bytes.")
print(s.recv())
```

If I run this, I get the following output: 
[image]

At this point, the terminal is kind of stuck and not working/responding anymore. To re run the script, i have to restart the kernel.
I'm using Python 3.7.9 and Spyder 4 (tried it with Python 3.8 before and downgraded the version after it wasn't working). 

Did you face that problem before and have any suggestions on how to solve it? Best regards!

pylessard, 2021-01-04:
Hi,
Unfortunately I have not expericened that issue.  Are you able to use ``isotpsend`` and ``isotprecv``?
Try to determine is the issue is in the driver or in the python layer.   If you are able to use the driver without Python then we may go deeper. Otherwise, if it's the driver, you may want to ask @hartkopp.

Regards.

kilianbr, 2021-01-04:
Thanks for your response. 

I tried using isotp with the following results:
[image]

First I sent a message using cansend (which works as expected). Using isotpsend it stucked again and did not response at all. So I assume the behavior is related to my driver? 

I already did the steps described here to install the module: https://github.com/hartkopp/can-isotp.

Best regards

pylessard, 2021-01-04:
Yes, it is indeed quite strange. Normally, you need to run isotprecv at the same time to get a communication, but your candump is empty, which should not happen even without isotprecv running.

Maybe check your kernel log.

I would move this issue here : https://github.com/hartkopp/can-isotp to get help form the author of the driver.

Regards

kilianbr, 2021-01-05:
Ok, thanks for your help. I think I now know that it is a problem with my kernel. I opened a new issue and will close this one here.

Best regards

closed 2021-01-05 by kilianbr