pylessard/python-can-isotp #39: Receiving from multiple ECUs using single bus

dpatel20, 2021-01-13:
I wanted to check if my understanding is correct in what I am trying to do.

I want to send and receive to multiple ECUs (addresses). I am using python-can and udsoncan. 
I had assumed that I would instantiate one python-can bus and then just send up different stacks

```
bus = VectorBus(channel=0, bitrate=500000)
addr = isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x123, txid=0x456)
stack = isotp.CanStack(bus, address=addr)

addr2 = isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x567, txid=0x789)
stack2 = isotp.CanStack(bus, address=addr2)
```

However, I find that no messages are received. If I create a ```bus2``` and use that in ```stack2```, it works fine. But is it correct that I need to instantiate a new bus for each address (ECU) that I want to communicate with?

I understand python-can will notify as required (i.e. for the messages (arbitration IDs) you are interested in) so I thought it was possible for only one bus to be used. Or is python-can-isotp consuming all messages received on that bus?

pylessard, 2021-01-13:
Hi,
This question has been asked few times. 

Make 2 bus objects and put filters on them so that one stack does not consume the message of the other stack.

You can also make a dispatching mechanism where you read the message from the bus and then feed them to each stack. To do so, you need to write your own ``rxfn`` and ``txfn``.

Regards

dpatel20, 2021-01-13:
@pylessard 
The first option feels a little inefficient (having multiple busses) - do you agree?
For the second option, I am willing to give it a go but I am unclear how this should work. I have seen the docs about defining your rx and tx functions but how can they dispatch to different stacks? 

I assume by stack here we mean ```isotp.TransportLayer```. If I create multiple ```isotp.TransportLayer``` (one per ECU), each will run its own version of the tx an rx function, each doing their own tx and rx (e.g. ```bus.recv()```) on the same bus.

I think I am not understanding what you mean - could you please add a bit more detail or some pseudo-code?

pylessard, 2021-01-14:
Hi @dpatel,
I don't see any issues in having 2 bus object. I f there is, I'd be willing to know them. The thing is that you must prevent one TransportLayer to empty the receive buffer from the message meant to the other TranportLayer. Having 2 buses object will give you 2 independent receive buffer and setting filters will correctly dispatch the CAN messages to the right receive buffer.

The dispatching mechanism, like I propose it would roughly go as follow :
 - Make 2 Queue object.
 - Make a thread that read the bus object.
 - For each massage that you read, check if it is meant to one of the transport layer. You can use the ``Address.is_for_me()`` method for that.
 - Put the message in the right queue.  
 - Write a ``rxfn`` for each ``TransportLayer`` function that will read the correct Queue. You can use Partial functions to avoid writing 2 times the same code.

Hope that helps.

dpatel20, 2021-01-14:
That's great. Thank you for your help. I agree that having a different bus with filters for each address is the easier way to go. My only concern was instantiating lots of python-can buses could causes issues or CPU load. I assume each instantiation of ```bus``` means another hook into the hardware and each one doing it's own receive, etc.

Whereas your second option of single thread doing all receiving and then just placing on appropriate queues would be more efficient (potentially...at least in my mind :) ).

pylessard, 2021-01-15:
Shall I close this issue then?

closed 2021-01-16 by dpatel20

feuerball, 2023-02-02:
Sorry to dig out this old topic, but I think there is a problem with multiple busses.

We are doing a broadcast through a larger database with a subprocess for each ecu, which results in more than 64 busses at the same time.
vxlapi only supports 64 open channels in parallel, thats why we limit the maximum open channels to this and wait for other subprocesses to be finished.

In general we would only need one bus for each channel of the vxlapi