pylessard/python-can-isotp #40: Receive flow control frames from ECU when transmitting, rather than can-isotp receiver?

davidva-ecco, 2021-02-24:
Thank you so much for making this module.

I am trying to use the isotp module to send data to an ECU on a Vector CAN interface, using the transmit example from the documentation.  In the bus trace, I see the First Frame from can-isotp, and a response Flow Control from my ECU.  However, the library then reports that it timed out while waiting for a flow control frame.

I can get can-isotp to communicate with itself by running the transmit example and threaded receive example code side by side, but it still appears can-isotp is generating the flow control frame and the transmitter is using that (trace from CANalyzer, ECU FC in green and can-isotp FC in yellow):

[image]

Can can-isotp be configured to use Flow Control frames from an ECU rather than generating them in the can-isotp receiver side?  Any tips on how to accomplish that?

pylessard, 2021-02-24:
Hi,
Something is wrong in your setup. The isotp stack shouldn't send a Flow Control if it has initiated the communication. Do you have multiple ECU talking on the same IDs? Does your stack have different Rx and Tx addresses?

Maybe share your configuration?

davidva-ecco, 2021-02-24:
I have a custom ECU connected to channel 1 of the VN1610 Vector interface (channel '0' in Python) running at 500kbps.  It expects requests from normal physical address 0x711 and responds with address 0x710.  I'll focus on the transmit-only code I'm trying to use since based on your reply it sounds like that should handle incoming Flow Control frames from my ECU.  Here's that code, which should be the same as the example code from the docs except for the use of can.Bus() instead of VectorBus():

```
import can
import isotp
import logging
import time

def my_error_handler(error):
   logging.warning('IsoTp error happened : %s - %s' % (error.__class__.__name__, str(error)))

bus = can.Bus(interface='vector', channel='0', bitrate='500000', receive_own_messages=True)
addr = isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x710, txid=0x711)
#layer = isotp.TransportLayer(rxfn=my_rxfn, txfn=my_txfn, address=addr)
stack = isotp.CanStack(bus, address=addr, error_handler=my_error_handler)

#bus.send(can.Message(arbitration_id=0x710, channel='0', dlc=8, data=[1,2,3,4,5,6,7,8]))

print("Sending payload.")
stack.send(b'Hello, this is a long payload sent in small chunks')
while stack.transmitting():
   stack.process()
   time.sleep(stack.sleep_time())

print("Payload transmission done.")
```

When I run that:

```
$> python .\IsoNetworkLayer.py
Sending payload.
WARNING:root:IsoTp error happened : FlowControlTimeoutError - Reception of FlowControl timed out. Stopping transmission
WARNING:isotp:Reception of FlowControl timed out. Stopping transmission
Payload transmission done.
```

And the trace from the CANalyzer software:

[image]

Perhaps I need to either send a 3 byte FC frame or enable padding in can-isotp?

davidva-ecco, 2021-02-24:
I figured it out, my ECU was setting the CAN extended ID bit even though it shouldn't have been.  Traced it down to `_is_for_me_normal()` in address.py and found out `if self.is_29bits == msg.is_extended_id` was evaluating to False.

Thanks for helping me find and fix a firmware bug ;-)  I appreciated your prompt response!

closed 2021-02-24 by davidva-ecco

pylessard, 2021-02-25:
Ah! We can indeed see the 710x in your canalyzer screenshot. Glad it helps!