pylessard/python-can-isotp #41: ConsecutiveFrameTimeoutError for first consecutive frame

MaratSabitov, 2021-03-12:
I found that ConsecutiveFrameTimeoutError was not raised, although, the first consecutive frame was not received after first frame and flow control exchange:
S                                 R
-----First Frame----->  
<--- Flow Control----

Is that expected behaviour - that ConsecutiveFrameTimeoutError raises only if consecutive frame receieved with delay.
What if the first consecutive frame will not come at all? 
I thought in this case ConsecutiveFrameTimeoutError should be raised.

pylessard, 2021-03-12:
Hi, the error should raise. It's even covered by a test case
https://github.com/pylessard/python-can-isotp/blob/ecbf5c8150fc8c9266b5898deb41c8e8beb91bde/test/test_transport_layer.py#L238-L250

Do you have more information on this? Can log? stack configuration? Code sample?

MaratSabitov, 2021-03-12:
Thanks for your quick answer.

Yeah, i saw this test.
And i mean that it should be raised even without incoming CF with index=1 ( line 246),
because stack received first frame, sent flow control and waits consecutive frame/frames, but  it was not received on time.

In this case timer_rx_cf is already expired.
And this is checked in process_rx().
But the problem is that this method is called only if message received.
So, timer expired - but need incoming message to trigger the check and raise the exception.

So, I just have doubts that this is up to the standard (Network layer timing part),
that's why i am asking this question.

pylessard, 2021-03-12:
You seem to be right.  I had no unit test that tested that specific case.  I will look into this.

closed 2021-03-12 by pylessard

MaratSabitov, 2021-03-13:
Yeah, i see.
Yesterday i had some troubles with comments.
Thanks.

pylessard, 2022-03-06:
This wasn't completely fixed. 
Issue still exist and logged with #63