pylessard/python-can-isotp #42: FlowControl are not read by the stack most of the times

sylar89ash, 2021-04-04:
I have can-isotp module running on RaspeberryPi4 and I am trying to send a multi-frame data from CAN using CAN HAT.
But whenever the module is sending the First frame it keeps on waiting for FC till it times out. Even though the Flow control frame was received on the Bus. I could verify that by printing the frame received also in candump command.

Module somehow doesnt recognize the FC frame and gives the timeout error.

Please guide me in this, my project execution has got stuck due to this issue.

pylessard, 2021-04-04:
Hi, it's a bit hard to figure what's going on without seeing your code.

My guess is that you have another task that reads the bus and empty the receive buffer, therefore the stack doesn't receive the message.

Otherwise it could beba misconfiguration of addresses.

For more help, share a can log and your code.

Regards

sylar89ash, 2021-04-05:
Hi, thank you for this info, but this attached image will show how I am talking about missing flow control frames, will share the code soon.
Thanks 
[image]

sylar89ash, 2021-04-05:
Hi, Yes you were right, there was a code line which was reading the bus and emptying the buffers, but as requirement I want to be able to read some responses and do some specific task as well. 
Is there any way I can do both? Read the bus as well as dont skip or miss the Flow control frame.

Thanks

pylessard, 2021-04-05:
Hi, I think there is at least 5 issues asking this question :)
You can make a dispatching mechanism that reads the bus then copy the message if necessary. If you do so, you will have to writre your own txfn and rxfn callbacks. 

Otherwise you can have multiple bus instance that have filters on them to select the messages to read.

Since this issue keep coming back, I will document the problem and maybe write something to work it around.

Regards

pylessard, 2021-04-05:
Oh, and since you are on linux, you may want to consider IsoTP socket. They perform better.

Regards

sylar89ash, 2021-04-06:
Ok, will give a go.
But recently, I have been receiving alot of following messages won CLI 

**Received invalid CAN frame. Received message with unknown frame type 15**

and the after certain number of trials, I get,

    **can_isotp_tx(mcan_bus, mtx_IsExtended, mtx_RxCANID, mtx_TxCANID, bytearray(mtx_Data))
ValueError: byte must be in range(0, 256)**

can you advice why this behavior?

pylessard, 2021-04-06:
Hi,
It looks like you feed non-isotp data to the iso-tp stack. Good chance that your address is not well defined and that the stack believe this non iso-tp traffic is meant to be parsed.

As for the second error, I don't really understand if the error is raised by the stack or the code you wrote. I guess you try to convert into bytearray something that isn't convertible.

sylar89ash, 2021-04-06:
Hi,
The channel we are using communicates on same CANIDs both for ISOTP messages as well as RAW CAN Messages. And the warning appearing is not an issue, they are delaying my Multi-frame transmission by 3-4 Seconds. Is this warning appear only for Rx messages? If yes, how can we keep them away or filter them out?

the second error comes to the data variable, where the isotp data is fed. the data is being sent is same, but the somehow the exception is thrown from messages.py from the library code.

pylessard, 2021-04-06:
You need a way to separate the ISOTP traffic from the rest of the traffic. That's the goal of the address.  If you don't respect the address, your communication will be unreliable.

for the second error, I don't know what Message.py is. you have to share more details if you want help.

closed 2021-04-17 by pylessard