pylessard/python-can-isotp #43: stack.process() got stuck during milti frame Tx

sylar89ash, 2021-04-20:
Hi,

Observed some weird behavior in my Python code for transmitting Multi-frame CAN MESSAGE.

code: 
While stack.transmitting():
    Stack.process()
    time.sleep(stack.sleep_time())

the execution hangs in stack.process() function call and eventually the while loop doesn’t break causing the whole code stuck.

please advise.

pylessard, 2021-04-20:
How is your ram usage when that happens?
Possibly the internal queue that is full

sylar89ash, 2021-04-22:
No, as it is quite randomly observed, I have not taken any note of RAM usage, will check it now.
And the internal queue must not be full as I have faced this issue just after booting the system and running the code for first time.

pylessard, 2021-05-03:
Any update on this?

sylar89ash, 2021-05-04:
Ya, so I checked the RAM usage as 23% and the queue is supposed to be empty as this happened in the first run itself.

pylessard, 2021-05-04:
How do you ensure that the faulty step is `stack.process()`
How did you define your rxfn and txfn?

I need some input from you. stack.process is non-blocking.

pylessard, 2021-05-13:
I am closing this issue as there is too few details to conclude that there really is an issue to be investigated.

closed 2021-05-13 by pylessard