pylessard/python-can-isotp #44: Extended address modes don't receive anything

MaratSabitov, 2021-05-04:
Hi,
i tried to use stack with extended addressing modes (11 bits, 29 bits) - send works,
but it doesn't receive anything.

I think there is a bug in Address._is_for_me_extended():
line 228:  return msg.arbitration_id == self.rxid and int(msg.data[0]) == self.source_address

May be it should be self.target_address instead of self.source_address?

pylessard, 2021-05-04:
Source address is the address of your device. So other devices needs to sends to this address to rech your client
You may have inverted your addresses.

For more help, share some code.

Regars

MaratSabitov, 2021-05-04:
I mean, that according to specification the first byte of payload contains "network target address".
Moreover, Address.validate() for AddressingMode.Extended_11bits, AddressingMode.Extended_29bits checks (line 123):
    self.target_address is None or self.rxid is None or self.txid is None

pylessard, 2021-05-04:
For a remote device that wants to talk to your client, your client source address is the device target address.
I'm not sure what point you try to make with your second statement as you point out a line of code without additional information.

If you want help with your issue, share some code, can log or standard references.

MaratSabitov, 2021-05-05:
Sorry,
it was my misunderstanding.
I didn't provide source_address value for Address but validation passes in that case - it checks rx_id, tx_id and target_address only.
So, i think issue can be closed.

pylessard, 2021-05-05:
Indeed, it looks like address validation may not enforce a source address when it should. I will review this.

closed 2021-05-09 by pylessard