pylessard/python-can-isotp #47: stmin is not consistent with multiframe messages

Yodawgz0, 2021-07-02:
while using isotp.CanStack & stack.process and setting stmin in funtion set_fc_opts doesn't have a consistent stmin time gap between messages of flow control. if 10 ms is set then there are instances of 0.008 or 0.006 which occurs during runtime with a complimentary stmin of 0.011 or 0.013 . 

File: 1--------

`import isotp
import can
from can.interface import Bus
import time
while True:
    isotp_params = {
     'stmin': 255, # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
     'blocksize' : 0, # Request the sender to send 8 consecutives frames before sending a new flow control message
     'wftmax': 0, # Number of wait frame allowed before triggering an error
     'tx_data_length': 8, # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
     'tx_padding': 0, # Will pad all transmitted CAN messages with byte 0x00. None means no padding
     'rx_flowcontrol_timeout': 1000, # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
     'rx_consecutive_frame_timeout': 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
     'squash_stmin_requirement': False # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
    }
    #lst = [40, 41, 42, 34, 48, 50, 63, 15, 66, 99, 12, 87, 59, 9, 78, 44]
    lst = []
    data = bytes(lst)
    bus = Bus(bustype='socketcan', channel='can1', bitrate=500000, receive_own_messages=True)
    rxid = 0x628
    txid = 0x6A8
    isExtId = False
    s = isotp.socket()
    tp_addr = isotp.Address(isExtId, rxid=rxid, txid=txid)
    stack = isotp.CanStack(bus, address = tp_addr, params = isotp_params)
    s.set_fc_opts(stmin=10, bs=0)
    s.bind("can1", tp_addr)
    
    stack.send(data)
    while stack.transmitting():
        stack.process()
        time.sleep(stack.sleep_time())
    time.sleep(3)
`

FIle 2 ---------

`import isotp
import can
from can.interface import Bus
import time

isotp_params = {
     'stmin': 255, # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
     'blocksize' : 0, # Request the sender to send 8 consecutives frames before sending a new flow control message
     'wftmax': 0, # Number of wait frame allowed before triggering an error
     'll_data_length': 8, # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
     'tx_padding': 0, # Will pad all transmitted CAN messages with byte 0x00. None means no padding
     'rx_flowcontrol_timeout': 1000, # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
     'rx_consecutive_frame_timeout': 1000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
     'squash_stmin_requirement': False # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
    }

rxid = 0x6A8
txid = 0x628
isExtId = False

bus = Bus(bustype='socketcan', channel='can0', bitrate=500000, receive_own_messages=True)

lst = [10, 21, 72, 33, 48, 50, 63, 15, 66, 99, 12, 87, 59, 9, 78, 44, 10, 21, 72, 33, 48, 50, 63, 15, 66, 99, 12, 87, 59, 9, 78, 44]
for x in range(190):
     lst.append(x)
data = bytes(lst)

def can_isotp_tx(bus, isExtId, rxid, txid, data, receiver = False): 
     s = isotp.socket()   
     tp_addr = isotp.Address(isExtId, rxid=rxid, txid=txid)
     s.bind("can0", tp_addr)
     stack = isotp.CanStack(bus = bus, address = tp_addr, params = isotp_params)
     s.send(data)
     time.sleep(2)
while True:
     can_isotp_tx(bus, isExtId, rxid, txid, data, receiver = False)

`

Also Can you clear that what is the difference between 
s = isotp.socket() & s.set_fc_opts & s.send

and 

s= isotp.CanStack(params) & s.process

which would be better in order to handle proper flow control

pylessard, 2021-07-02:
The answer to your questions are all in the documentation.
TransportLayer (canstack) is a python implementation that runs in the user space.
Socket uses a Kernel module.

Main difference is timing respect as you pointed out.

Also, the TransportLayer object bases its timing on the Python `time` module. Stmin will be respected but if your driver buffers data than flush many message quickly, there isn't much I can do unless there's a handle in the user space I can use.

You can share CAN log to show your issue.

Finally, in your first file, you instantiate a CanStack and a isotp socket. Its one or the other, not both. And in file2 you open a socket for each datagram, you should keep it open.

Yodawgz0, 2021-07-05:
When using socket with set_fc_opts(stmin = 10) and only using isotp socket

File1-->>

```
import isotp
import can
import time

rxid = 0x6A8
txid = 0x628
isExtId = False

lst = []
for x in range(40):
     lst.append(x)
     
data = bytes(lst)

def can_isotp_tx(isExtId, rxid, txid, data): 
     s = isotp.socket()
     tp_addr = isotp.Address(isExtId, rxid=rxid, txid=txid)
     s.bind("can0", tp_addr)
     s.send(data)
     time.sleep(2)
while True:
     can_isotp_tx(isExtId, rxid, txid, data)

```
File2 -->>

```
import isotp
import can
import time

rxid = 0x628
txid = 0x6A8
isExtId = False

while True:
    s = isotp.socket()
    tp_addr = isotp.Address(isExtId, rxid=rxid, txid=txid)
    s.set_fc_opts(stmin=10, bs=0)
    s.bind("can1", tp_addr)
    time.sleep(3)

```

When having 10 ms as Stmin I had occurrences where 0.08 is there very rarely 0.07 any idea why this is happening?

Log1 -> 

```
 (001.950731)  can1  628   [8]  10 28 00 01 02 03 04 05
 (000.001082)  can1  6A8   [3]  30 00 0A
 (000.013336)  can1  628   [8]  21 06 07 08 09 0A 0B 0C <---
 (000.007335)  can1  628   [8]  22 0D 0E 0F 10 11 12 13 <---
 (000.010028)  can1  628   [8]  23 14 15 16 17 18 19 1A
 (000.010034)  can1  628   [8]  24 1B 1C 1D 1E 1F 20 21
 (000.010069)  can1  628   [7]  25 22 23 24 25 26 27
```

Log2 -> 
```
 (001.950988)  can1  628   [8]  10 28 00 01 02 03 04 05
 (000.000717)  can1  6A8   [3]  30 00 0A
 (000.011121)  can1  628   [8]  21 06 07 08 09 0A 0B 0C
 (000.009958)  can1  628   [8]  22 0D 0E 0F 10 11 12 13
 (000.011699)  can1  628   [8]  23 14 15 16 17 18 19 1A  <---
 (000.008309)  can1  628   [8]  24 1B 1C 1D 1E 1F 20 21  <---
 (000.010079)  can1  628   [7]  25 22 23 24 25 26 27
```

hartkopp, 2021-07-05:
The isotp kernel driver sends the CAN frames very accurate every 10ms.
I assume the jitter is introduced in CAN driver level.
What CAN hardware are using?
Can you post the result of `ip -det link show can0` ?

Yodawgz0, 2021-07-05:
```
5: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 65536
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```
is this what you need @hartkopp ?

hartkopp, 2021-07-05:
I checked the isotp stack with the virtual CAN:
Terminal 1:
`candump -td vcan0`

Terminal 2:
`isotprecv -s 321 -d 123 vcan0 -m 0x0A`

Terminal 3:
`isotpsend -s 123 -d 321 -D 30 vcan0`

This produces in Terminal 1:
```
 (019.727433)  vcan0  123   [8]  10 1E 01 02 03 04 05 06
 (000.000024)  vcan0  321   [3]  30 00 0A
 (000.010053)  vcan0  123   [8]  21 07 08 09 0A 0B 0C 0D
 (000.010087)  vcan0  123   [8]  22 0E 0F 10 11 12 13 14
 (000.010057)  vcan0  123   [8]  23 15 16 17 18 19 1A 1B
 (000.010128)  vcan0  123   [4]  24 1C 1D 1E
```

So I strongly assume this ~3ms delay for the first consecutive frame is introduced somewhere in the driver layer.

What's your current Kernel version on the RasPi? `uname -a` ??

Yodawgz0, 2021-07-06:
The current Kernel Version on RasPi is 

`Linux raspberrypi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux`

hartkopp, 2021-07-06:
As you wrote you are using this hardware:
https://www.waveshare.com/2-ch-can-hat.htm

Did you follow this documentation:
https://www.waveshare.com/wiki/2-CH_CAN_HAT#Install_libraries
and installed that _bcm2835 library_ ??

And which RasPi version do you use?

I assume installing that bcm2835 library introduces the problem, as the latest 5.10 kernel already has the best SPI support AFAIK.

Yodawgz0, 2021-07-06:
I don't think I had installed *bcm2835 library* explicitly also I'm not able to find any folder named as such in `home/pi`

The version of RasPi -> 4

Also , using `isotpsend`  I tried sending messages (There were multiple iterations involves since continuous tx is there), instead of using `socket()` and `send `

```
data = bytes([1,2,3,4,5,6,7,8,9])
os.system("echo {} | isotpsend -s 123 -d 321 can0".format(data))
```
[image]

Any suggestion ? on what I can do that would reduce number of discrepancies.

hartkopp, 2021-07-07:
IMO the problem can be located on the SPI/MCP2515 driver and we should probably close this issue in this python-can-isotp thread.
I assume with `cangen -g10 can0` you  run into a similar jitter - even without isotp, right?
If so, I would suggest to continue the discussion on the Linux CAN mailing list ...

Yodawgz0, 2021-07-08:
Okay I will look into it.

Thank you soo much @hartkopp & @pylessard  for clearing doubts and helping me out. :) 

Closing the issue.

closed 2021-07-08 by Yodawgz0