pylessard/python-can-isotp #50: send flow control frame timeout

XXIN0, 2021-08-05:
I use PythonIsoTpConnection in udsoncan.connections to send and recive can msg,but when use it to test uds service 2E to write data, it will timeout sometimes when send flow control frame, please Look at the picture below
[image]

Hardware：vector cancasexl 
Python 3.7.9
 
here is code
class UdsLib:

    def set_can_config(self, tx_id=0x703, rx_id=0x70B):

        try:
            self.bus = can.interface.Bus(bustype='vector', channel='0,1', app_name='Python Can', bitrate=500000,
                                         receive_own_messages=True)
            self.tp_addr = isotp.Address(addressing_mode=isotp.AddressingMode.Normal_11bits, txid=tx_id, rxid=rx_id)
            tp_stack = isotp.CanStack(bus=self.bus, address=self.tp_addr, params=iso_tp_params)
            self.conn = PythonIsoTpConnection(tp_stack)
        except:
            AssertionError(sys.exc_info()[0])
        else:
            logger.info("Set Can config is ok!")

    def diagnostic_request(self, request_dat):
        request_pdu = binascii.a2b_hex(request_dat)
        if not self.conn.is_open():
            self.conn.open()
        try:
            self.conn.send(request_pdu)
        except can.CanError:
            AssertionError("Send request data fail")
        else:
            logger.info('Sending %d bytes : <%s>' % (len(request_pdu), binascii.hexlify(request_pdu)))

    def diagnostic_respond(self, respond_dat):
        expect_respond_pdu = binascii.a2b_hex(respond_dat)
        try:
            resp_pdu = self.conn.wait_frame(timeout=2)

            if resp_pdu is not None:
                while resp_pdu[0] == 0x7F and resp_pdu[2] == 0x78:
                    logger.info('Received %d bytes : <%s>' % (len(resp_pdu), binascii.hexlify(resp_pdu)))
                    resp_pdu = self.conn.wait_frame(timeout=3)
        except Exception as err:
            raise err

        if resp_pdu is not None:
            logger.info('ECU: Received %d bytes : <%s>' % (len(resp_pdu), binascii.hexlify(resp_pdu)))
            if resp_pdu.find(expect_respond_pdu) == 0:
                return True
            else:
                return binascii.b2a_hex(resp_pdu)
        return resp_pdu

    def close_can(self):
        self.conn.close()
        self.bus.stop_all_periodic_tasks()
        self.bus.shutdown()
        logger.info("close the device")

XXIN0, 2021-08-06:
CAN hardware: vector cancaseXL
Operating system: win10 21H1 19043.1110 x64
cumputer RAM:8G

it‘s very strange that this problem doesn't come up very often，I don't kown how to solve this problem, can you give some advice,or I can email you the code,thank you

pylessard, 2021-08-06:
Well, my advice is to profile the code and find where it get stuck for above 100ms.

I made my email public, its available in my profile, you cans end your code, I will have a quick look, but no guarantee. 
I will remove my email after you send your message.

Regards

XXIN0, 2021-08-07:
thank you,I've e-mailed the code to you

closed 2022-03-07 by pylessard