pylessard/python-can-isotp #52: How Sending with functional addressing in client

xinkunlong, 2021-08-12:
I am so sorry, I have been on a business trip before, so I have never time to reply

- why does it need to be functional?

1. I want to design a  software download tool for vehicle flashing, So I need to turn off the communication($28) enable and DTC detection($85) enable of other ECUs through functional addressing, and then refresh the ECU through physical addressing.

- do you expect an answer from your server?

1. No

- do you want to switch between functional and physical back and forth?

1. yes

for example
```code
client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession) # use functional addr
client.control_dtc_setting(services.ControlDTCSetting.SettingType.off)  # use functional addr
client.communication_control(services.CommunicationControl.ControlType.disableRxAndTx,0x01)   # use functional addr
client.change_session(services.DiagnosticSessionControl.Session.programmingSession)  # use physical addr
client.unlock_security_access(0x11) # use physical addr
                     ....
```

pylessard, 2021-08-12:
Hi, after reviewing this issue, I think it should be handled by the isotp layer.

What I propose is to add a configuration called ``default_target_address_type`` which will be used if no target_address_type is given to the ``send`` method. The UDS client does not specify this value, so the configuration would be used

If I take the example in documentation and your example, it would give something like this.

    bus = VectorBus(channel=0, bitrate=500000)                                          # Link Layer (CAN protocol)
    tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x123, rxid=0x456) # Network layer addressing scheme
    stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)               # Network/Transport layer (IsoTP protocol)
    conn = PythonIsoTpConnection(stack)                                                 # interface between Application and Transport layer
    with Client(conn, request_timeout=1) as client:                                     # Application layer (UDS protocol)

        stack.params.set('default_target_address_type', isotp.address.TargetAddressType.Functional)
        client.change_session(DiagnosticSessionControl.Session.extendedDiagnosticSession) 
        client.control_dtc_setting(services.ControlDTCSetting.SettingType.off)  
        client.communication_control(services.CommunicationControl.ControlType.disableRxAndTx,0x01)

        stack.params.set('default_target_address_type', isotp.address.TargetAddressType.Physical)
        client.change_session(services.DiagnosticSessionControl.Session.programmingSession)  # use physical addr
        client.unlock_security_access(0x11) # use physical addr

What do you think?

pylessard, 2021-09-01:
Just change the Address with `set_address`.
With normal addressing, physical or functional has no effect

closed 2022-02-19 by pylessard