pylessard/python-can-isotp #53: [CAN_FD] Padding disabled with tx_data_length set

4nthraxx, 2021-09-08:
Hi there,

I encountered the following behaviour:

The settings:
```
        self.bus = can.interface.Bus(bustype='vector', channel=0, bitrate=500000, app_name="CANoe", fd=True,
                                     data_bitrate=2000000, bitrate_switch = True)
        addr = isotp.Address(isotp.AddressingMode.Normal_11bits, rxid=0x604, txid=0x704, is_fd=True)
        self.stack = isotp.CanStack(self.bus, address=addr, error_handler=self.my_error_handler,

                                    params={'tx_padding': 0, 'can_fd': True,
                                            'bitrate_switch': True,
                                            'tx_data_length': 64})
```

If the tx_data_length is not there the CAN_FD fraems are padded to 8Bytes.

As soon as the 'tx_data_length': 64 is set the padding is no longer working and we're sending "real" frame lengths again.

Is this intended? Is it possible to setup the padding beaviour to always pad to 8 Bytes (or a configurable value?).

Am I missing something? Is this a feature or a bug? ;)

Thanks in advance!

/A

pylessard, 2021-09-08:
I don't exactly remember what the standard says, i will have to double check, but you probably wants to use another parameter named tx_data_min_length

https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#tx_data_min_length

4nthraxx, 2021-09-08:
It is : tx_data_min_length(int) indeed. It helped instantly.

Thanks a lot!

closed 2021-09-08 by 4nthraxx