pylessard/python-can-isotp #54: Python-can-isotp consume a asc file

Kheffache, 2021-11-05:
Hi, i want to know if the Python-can-isotp can consume a CAN log file (.asc .txt ...) to extract the payload of the frame that are CAN TP data frames. If not any help on a possible solution ?

hartkopp, 2021-11-05:
You can simply use the isotpsniffer from the can-utils https://github.com/linux-can/can-utils package.
Convert the asc to a log file (with asc2log) then replay the logfile to a virtual can (e.g. canplayer -I logfile.log vcan0=can0) and then run isotpsniffer on that vcan0 with the needed CAN-IDs to let the Linux kernel re-assemble the ISO-TP PDUs and print them.

Kheffache, 2021-11-08:
> You can simply use the isotpsniffer from the can-utils https://github.com/linux-can/can-utils package. Convert the asc to a log file (with asc2log) then replay the logfile to a virtual can (e.g. [...]

Thanks for your answer, but can this tool work under windows ?

hartkopp, 2021-11-08:
Ehm - no. It uses the isotp implementation inside the Linux kernel. Sorry - was not part of the question ;-)

pylessard, 2021-12-28:
Can we close this issue?

closed 2022-01-24 by pylessard