pylessard/python-can-isotp #55: Register callback per addressing; multi-threading

larsrde, 2021-12-21:
Hi,
I have to deal with a rather complex ecu, that sends different PDUs (on 15 different Rx/Tx addresses) at an arbitrary time and in random order.
I would like to handle them all with a single python script. Hence I am wondering if a mechanism that accepts a simple address-to-callback dictionary might do it. All the can stack handling and threading is hidden from my program. The callback methods are provided with the received message and voila.

However for today, my simple question is, is it safe to run 15 threads (as described here https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#threaded-reception-with-python-can) each with different addressing.

If I misunderstood the whole concept, then please let me know, too  :-).

Thanks
Lars

pylessard, 2021-12-21:
Hi,
Yes, you can do something like that. But you can also put the 15 stacks object in the same threads. The ``process`` method is non-blocking. So a simple for loop on your 15 objects should do it.

    min_sleep_time = 0.05
    for stack i stack_list:
        stack.process()
        min_sleep_time = min(stack.sleep_time())
    time.sleep(min_sleep_time )

Also, make sure to put filters on your Bus object so that one bus does not consume the emssage of the other bus.
if it cause troubles, you can always make a dispatching mechanism where you read the CAN message from a single Bus object, then redirect the message to a specific queue.  You can then write your own ``rxfn`` function that will read from that queue.

Hope that help a bit

pylessard, 2021-12-28:
Do you meed anything else? Shall I close the issue?

larsrde, 2021-12-29:
Thanks for your clarification. Will give it a try.

closed 2021-12-29 by larsrde