pylessard/python-can-isotp #68: Unstable Connection

blackmaktj, 2022-04-22:
Hi, I have two problems with the library. 

1- The first one is a bit weird. When my laptop is connected to the dock station and other devices I get a fellow control error. But when I disconnect the laptop it works properly and I can have communication through CAN. The docking station is connected to two monitors. I have tried several times but I always the same results. I am not sure, but I think STmin is set to 12ms in the embedded board. If it makes sense in this case I can ask for the embedded device specification. Is there any solution to this problem? 

2- The second problem: I have a file that I want to write to the board. The payload of the file is 280 bytes. The defined Blocksize on the board is set to 10. The problem is that on the embedded board it is defined that in the first frame it should be defined that the length of this data is 280 bytes and after 10 frames the "CONSECUTIVE FRAME" will be called, until the last frame. If I set the Blocksize on the embedded device to ex. 200 the problem will be solved because it sends all the frames without calling "CONSECUTIVE FRAME" but in this case, the default is 10. And if I shrink the package, because the length of the data changes, the data is not valid to the board. Does anyone have a solution? 

If there is a need for the sample code please let me know.
Thank you in advance :)

blackmaktj, 2022-04-23:
1- I am not sure that is a grounding problem. Because everything works file with CAN vector software. But when I send data ( using python ) to the board I get follow control time out error. When the board is in debugger session it shows fellow control time-out error in the debugger as well. Can the PC configuration have an effect on the connection? I assume that my laptop can not handle two extra monitors, being connected to a docking station with an open chrome browser and some other software. Can it be? by the way I will try with another laptop on Monday!

2- Sorry for my bad explanation. I have attached the log file. In the first log file as you can see I set the block size on the board(embedded device) to e.q 200. And transferring data is working fine. But if you lock at the second log file you see if I set the block size on the board to 10 (which is default and should be) I get a FLowControl time out error. Stopping transmission. Also, I set the Blocksize parameters in my CANtp config file to 10 but I got the same error.

First_Log:
`13:31:53,941 Diag INFO The Packet is : <b'0010000001000a010000200000000001020c00019d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d02000845b604089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200085da904089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d0200089d020008e9b6040839b7040899b6040889b70408d9b7040829b804089d0200089d0200089d0200080000000000000000000000000000000000000000000000009d020008000000009d0200089d020008000000009d0200089d020008000000000000000000000000a8cf'>  with the length of  278 :
13:31:53,943 isotp DEBUG Sending : <601> (8) b'1116001000000100'
13:31:53,968 isotp DEBUG Receiving : <681> (8) b'30c8010000000000'
13:31:53,968 isotp DEBUG Sending : <601> (8) b'210a010000200000'
13:31:53,968 isotp DEBUG Sending : <601> (8) b'22000001020c0001'
13:31:53,970 isotp DEBUG Sending : <601> (8) b'239d0200089d0200'
13:31:53,970 isotp DEBUG Sending : <601> (8) b'24089d0200089d02'
13:31:53,971 isotp DEBUG Sending : <601> (8) b'2500089d0200089d'
13:31:53,971 isotp DEBUG Sending : <601> (8) b'260200089d020008'
13:31:53,972 isotp DEBUG Sending : <601> (8) b'279d0200089d0200'
13:31:53,972 isotp DEBUG Sending : <601> (8) b'280845b604089d02'
13:31:53,972 isotp DEBUG Sending : <601> (8) b'2900089d0200089d'
13:31:53,972 isotp DEBUG Sending : <601> (8) b'2a0200089d020008'
13:31:53,972 isotp DEBUG Sending : <601> (8) b'2b9d0200089d0200'
13:31:53,973 isotp DEBUG Sending : <601> (8) b'2c089d0200089d02'
13:31:53,973 isotp DEBUG Sending : <601> (8) b'2d00089d0200089d'
13:31:53,973 isotp DEBUG Sending : <601> (8) b'2e0200089d020008'
13:31:53,974 isotp DEBUG Sending : <601> (8) b'2f9d0200089d0200'
13:31:53,975 isotp DEBUG Sending : <601> (8) b'20089d0200089d02'
13:31:53,976 isotp DEBUG Sending : <601> (8) b'2100089d0200089d'
13:31:53,977 isotp DEBUG Sending : <601> (8) b'220200089d020008'
13:31:53,977 isotp DEBUG Sending : <601> (8) b'235da904089d0200'
13:31:53,977 isotp DEBUG Sending : <601> (8) b'24089d0200089d02'
13:31:53,977 isotp DEBUG Sending : <601> (8) b'2500089d0200089d'
13:31:53,979 isotp DEBUG Sending : <601> (8) b'260200089d020008'
13:31:53,980 isotp DEBUG Sending : <601> (8) b'279d0200089d0200'
13:31:53,980 isotp DEBUG Sending : <601> (8) b'28089d0200089d02'
13:31:53,980 isotp DEBUG Sending : <601> (8) b'290008e9b6040839'
13:31:53,980 isotp DEBUG Sending : <601> (8) b'2ab7040899b60408'
13:31:53,985 isotp DEBUG Sending : <601> (8) b'2b89b70408d9b704'
13:31:53,987 isotp DEBUG Sending : <601> (8) b'2c0829b804089d02'
13:31:53,988 isotp DEBUG Sending : <601> (8) b'2d00089d0200089d'
13:31:53,988 isotp DEBUG Sending : <601> (8) b'2e02000800000000'
13:31:53,990 isotp DEBUG Sending : <601> (8) b'2f00000000000000'
13:31:53,990 isotp DEBUG Sending : <601> (8) b'2000000000000000'
13:31:53,990 isotp DEBUG Sending : <601> (8) b'210000000000009d'
13:31:53,991 isotp DEBUG Sending : <601> (8) b'2202000800000000'
13:31:53,992 isotp DEBUG Sending : <601> (8) b'239d0200089d0200'
13:31:53,992 isotp DEBUG Sending : <601> (8) b'2408000000009d02'
13:31:53,993 isotp DEBUG Sending : <601> (8) b'2500089d02000800'
13:31:53,994 isotp DEBUG Sending : <601> (8) b'2600000000000000'
13:31:53,994 isotp DEBUG Sending : <601> (7) b'2700000000a8cf'
13:31:54,53 Diag INFO Send Time: 0:00:03.269666
13:31:54,977 isotp DEBUG Receiving : <681> (8) b'1010001000000100'
13:31:54,978 isotp DEBUG Sending : <601> (3) b'304020'
13:31:55,12 isotp DEBUG Receiving : <681> (8) b'2104000000210003'
13:31:55,45 isotp DEBUG Receiving : <681> (8) b'2200356200000000'
13:31:55,55 Diag INFO Board send 16 bytes : <b'00100000010004000000210003003562'>

13:31:55,70 Diag INFO The Packet is : <b'0010000002000a010000200000000002020c0001dff8a8d02a4880472a4c2b4dac4209da21686268a368043ba2bfc858d050fae70c34f3e7254b264ca34208da19685a680020043aa4bf8850fbe70833f4e721482149016021488047acd5040c0008002000000000acd5040c002000209806000044dc040c008000205403000000c0ff1f000000000000002000080000982600202c16000054830020ac1c000020080020e017000000a000200048000000e800200000000000e800200000000000080020a9a50408480202086c0202086c020208ac0202080000020808ed00e095a50408fee7000080b586b000af78607b68002b0ad14ff0f00383f31188bff36f8fbff34f8f3b6100bffee71823bb60bb68182b77d1'>  with the length of  278 :
13:31:55,71 isotp DEBUG Sending : <601> (8) b'1116001000000200'
13:31:55,119 isotp DEBUG Receiving : <681> (8) b'30c8010000000000'
[... cut ...]
11:46:47,768 isotp DEBUG Sending : <601> (8) b'2af00afffb697b63'
11:46:48,773 isotp WARNING Reception of FlowControl timed out. Stopping transmission
11:46:48,838 Diag INFO Send Time: 0:00:12.485033
11:46:48,859 isotp DEBUG Receiving : <681> (8) b'1010001000000600'
11:46:48,859 isotp DEBUG Sending : <601> (8) b'3000200000000000'
11:46:48,893 isotp DEBUG Receiving : <681> (8) b'2104000000210001'
11:46:48,925 isotp DEBUG Receiving : <681> (8) b'2200ad1800000000'
11:46:49,349 Diag INFO Board send 16 bytes : <b'0010000006000400000021000100ad18'>`

pylessard, 2022-04-24:
Haa, "fellow control timeout"=flowcontrol timeout. It is easier to understand now.

When you set block size to 10, your embedded board is expected ton send a flow control every 10 blocks and it doesnt. Most likely a problem with your embedded code

pylessard, 2022-05-15:
Shall we close this issue?

blackmaktj, 2022-05-22:
Just I will share how I did solve the problem. I figure out that when I close windows explorer the application work fine !! And after some message transmission, I could open windows explorer again and everything works fine. I have tried with different laptops and almost the same results and errors. But I do not have any reason for this problem. I run the application as a Real-Time priority but did not solve the problem. 

What solves my problem. I create the Bus object and then receive (read) some data from the board and then I create the Transport_Layer (Which I create to send/receive data through specific Rx and Tx channels) object to send my data and this works. 

Thank you.

closed 2022-07-07 by pylessard