pylessard/python-can-isotp #75: Connection with can

KC6320, 2022-08-26:
Hi @pylessard,

I have been using the entire stack ( udsoncan, can-isotp, python-can) and it is wonderful. Thanks.
This issue is somewhat related to https://github.com/pylessard/python-udsoncan/issues/121 that I had posted earlier. My team is planning to move some of the uds related testing on linux and decide to run the can and tp layer in kernel level. 

1) Using python-can with socket can as interface is just interfacing with socket CAN on linux I assume. If so, the isotp layer stills needs to be in python in user space. and this wont help much with the timing issue. Please correct me if wrong. 
2) Is the workflow below mentioned correct ? 
    a. load the third party isotp kernel module 
    b. use isotp.socket on python side
    c. Use IsoTPSocketConnection from udsoncan to create a connection. 
 Please correct me if wrong, 🤞
3) If I follow step 2. how should I connect the isotp to can layer at kernel level ? more importantly, how do I bring up physical can hardware connection ? I am sure the peak pcan drivers are already into the kernel. 

Awaiting your response for above 3 points. Thanks much.

pylessard, 2022-08-26:
1. the isotp kernel driver is built above socketcan. it handles the isotp layer in the kernel, not the userspace.

2. Yes you need to do what you described in 2. 

3. You configure your interface through your os using "ip link" command.  Just like you would configure your ethernet adapter. The isotp socket requires a device to work on, in that case, you should specify the name your os has given to your peak adapter  (something like "can0".  

Normally, after you plug you peak interface, you should have the interface created right away. If it is not, you might be missing drivers or socketcan support.

pylessard, 2022-08-26:
Check this : https://elinux.org/Bringing_CAN_interface_up

KC6320, 2022-08-29:
Thank you so much 👍. I will try it out

closed 2022-08-29 by KC6320