pylessard/python-can-isotp #80: Unrecoverable errors when rapidly sending new packets.

tlalexander, 2023-01-06:
Hello and thank you for your high quality library.

I am having an issue and I am not quite sure where to look for help. I have CAN communication working between a Raspberry Pi CM4 and an Arduino board running the iso-tp library. Communication works great, unless I transmit from the Raspberry pi too quickly. And then once I get an error, I have not been able to find any way to recover the program. However in a separate terminal on the raspberry pi I can use isotpsend to confirm that the link is fine. When my program gets in to the error state, it returns a timeout, and then subsequent attempts to send data, even with a long delay after the error, will throw the timeout again and transmission never succeeds again until I restart the program. Here is my test code which activates the error:

```
import isotp
import time
import traceback

s = isotp.socket()
s.bind("can1", isotp.Address(rxid=0x123, txid=0x456))

while True:
    try:
        s.send(bytearray(20))
    except Exception as e:
        traceback.print_exc()
    # If this sleep is 0.003 or less, I get errors.
    time.sleep(0.005)
```
I want to transmit data as quickly as possible, but I have been unable to find any way to check when it is safe to transmit again, and if I transmit too quickly I get in to the unrecoverable error state. As I say in my comment in the code above, if the sleep is set to 3ms instead of 5ms, the code will transmit several frames with success and then start throwing one of two errors.

Sometimes it will throw the following error, and then recover and continue sending more data:
```
Traceback (most recent call last):
  File "/home/acorn/acorn/vehicle/isotp_test.py", line 12, in <module>
    s.send(bytearray(20))
  File "/home/acorn/.local/lib/python3.9/site-packages/isotp/tpsock/__init__.py", line 84, in send
    return self._socket.send(*args, **kwargs)
OSError: [Errno 70] Communication error on send
```
In the test case I just ran, it hit that error three times, recovering each time, before throwing this error and never recovering again:
```
Traceback (most recent call last):
  File "/home/acorn/acorn/vehicle/isotp_test.py", line 12, in <module>
    s.send(bytearray(20))
  File "/home/acorn/.local/lib/python3.9/site-packages/isotp/tpsock/__init__.py", line 84, in send
    return self._socket.send(*args, **kwargs)
socket.timeout: timed out
```
My Arduino program is a tight loop that only waits for new frames and prints them to a terminal.
```
(omitting setup code)
void loop()
{
  if(CAN0.checkReceive() == CAN_MSGAVAIL)
  {
    if(isotp.receive(&rxMsg)==0)
    {
      isotp.print_buffer(rxMsg.rx_id, rxMsg.Buffer, rxMsg.len);
    }
  }
}
```
When my python program is in the unrecoverable error state, I am still able to send frames from another terminal on the raspberry pi with:
`isotpsend -d 123 -s 456 -D 90 -b can1`

And my Arduino prints the received frames in the terminal as expected.
This leaves me to believe there is a problem with the python socket.

I have tried to get the program to recover from this timeout error by closing the socket and opening it again, but it will hang when trying to close the socket. If I simply try to open it again without closing it, then it will hang at bind().

I have also tried to enable the WAIT_TX_DONE flag:
`s = isotp.socket(isotp.socket.flags.WAIT_TX_DONE)`
This code runs and changes the behavior. In this mode the error still occurs, but the code simply hangs on send, never throws the timeout error, and locks up the program. I am unable to close the program with CTRL-C and I have to kill the program from another terminal window.

This situation leaves me with several questions:
Firstly, it looks like what is happening under the hood is this is basically using a standard python socket, because I am not using the userspace driver. Raspberry pi includes the kernel driver. This has me wondering if I should even be asking here, because this is basically a standard python socket interacting with the iso-tp kernel driver. However I am running your library even if in my case it is not doing a whole lot, and it seems like you might know more about this than asking in a general python list of some sort.

As far as fixing this I have two possibilities: Find some way to check to make sure it is safe to send before sending more data, or find some way to recover from the error.

The weird thing is, the timeout is 100 milliseconds, and this data only takes 1 milliseconds to send, so when I enable WAIT_TX_DONE I would expect it to never fail. I don't understand what is happening when it locks up (when WAIT_TX_DONE is enabled) or when it constantly returns a socket.timeout and never recovers.

Any guidance on how to proceed would be appreciated. I may post a link to this issue in the isotp kernel module issues page as well to get more eyes. Either I am doing something wrong or there is a bug, and I suspect it is the former.

Thanks!

pylessard, 2023-01-06:
Hi,
That's interesting.  I will have a look at this in more details tomorrow (it's late for me).
At first glance, it looks like you put the [kernel driver](https://github.com/hartkopp/can-isotp) in a bad state. Maybe the author @hartkopp can give us more details.

I would look into the[ kernel driver code](https://github.com/hartkopp/can-isotp/blob/master/net/can/isotp.c) and also into the [python socket module](https://github.com/python/cpython/blob/main/Modules/socketmodule.c) to get more meaning of the error.

tlalexander, 2023-01-06:
Thank you! It is late for me too, but I will look at the code you linked tomorrow. Thanks for those references.

tlalexander, 2023-01-06:
Well I realized that I had completely incorrect use of the WAIT_TX_DONE flag, which should not be passed via the socket constructor. Since the flag is an int, it gets interpreted as a timeout so that only gave me a long timeout, which behaves like a lockup. Correct use of the flag with set_opts results in the behavior I want. The only remaining question is why the timeout behavior (with no flag) is unrecoverable. I posted more details to the other issue mentioned above.

tlalexander, 2023-01-14:
I've sorted my issues. It was incorrect use of the flag, combined with some behaviors which are basically considered expected, if confusing, when hitting errors when not using the flag. The remedy is: use the flag and use it properly.

closed 2023-01-14 by tlalexander