pylessard/python-can-isotp #82: Multiple ECUs without generating new Bus for every ecu

feuerball, 2023-02-05:
Hi there, I think there is indeed a problem with multiple busses.

We are doing a broadcast through a larger database with a subprocess for each ecu, which results in more than 64 busses at the same time.
vxlapi only supports 64 open channels in parallel, thats why we limit the maximum open channels to this and wait for other subprocesses finish

In general we would only need one bus for each channel of the vxlapi, so we don't need to "wait"..

pylessard, 2023-02-05:
You can overcome this by writin your own rxfn and txfn.  The one provided in the CanStack are convenience for easier integration, but if your use case is more complex, just write your own. it is very easy.

On the other hand, there is #76 that is open that I need to handle. I'm focusing on something else these days, but it's on the todo list.

feuerball, 2023-02-20:
Thanks for your answer. I was looking into the code, I might just quickfix it by myself.

Meanwhile some strange happened, but I don't understand why:

I create an VectorBus and CanStack for each ECU.
In our configuration we got the same ecu twice, but on different channels. Therefore i create two instances of bus & stack with the selected channels, but they both can hear each other. Why tho?

hartkopp, 2023-02-20:
I'm not really sure about the terminus "channel" and "bus" in the original question.

An ISOTP channel is something where TWO CAN IDs form a bidirectional channel for TWO ECUs. _**This channel is unique on a given CAN segment ("bus")**_. One ECU has e.g. 0x111 as TX ID and 0x222 as RX ID. This means exactly **_one_** other ECU (the other end of that channel)  has 0x111 as RX ID and 0x222 as TX ID. 0x111 and 0x222 are forbidden for any other ECUs on that bus.

As long as you follow this rule everything is fine - no matter how many "busses" you have.

feuerball, 2023-02-24:
Example:

I got two ECUS:

ECU 1
VX1161 Channel 1 
TXID 1
RXID 1

ECU2
VX1161  Channel 2
TXID 1
RXID 1

For each ecu i create a subprocess with seperate bus:

`
def can_canfd_connection(interface, interface_settings, channel, Request_ID, Response_ID, is_extended_frame):
    if interface.lower() == "vector":
        # because python start on 0 , vector hw config on 1
        interface_settings["channel"] = channel - 1
        interface_settings["app_name"] = "AutomaticDiagnostic"
        bus = VectorBus(**interface_settings)

    adressing_mode = isotp.AddressingMode.Normal_29bits if is_extended_frame else isotp.AddressingMode.Normal_11bits
    tp_addr = isotp.Address(adressing_mode, txid=Request_ID, rxid=Response_ID)

    # Network/Transport layer (IsoTP protocol)
    isotp_params = interface_settings.get("isotp_params")
    if isotp_params["tx_data_min_length"] == "None":
        isotp_params["tx_data_min_length"] = None

    stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)
    logger.debug(f"Returning isotp CanStack for {Request_ID}")

    return PythonIsoTpConnection(stack), bus`

Nethertheless, if i start the processing in parallel mode, bus 2 reacts on the rx from bus 1.

Hope its more clear now.

hartkopp, 2023-02-24:
> I got two ecus:
> 
> **ECU 1** VX1161 Channel 1 TXID 1 RXID 1
> 
> **ECU2** VX1161 Channel 2 TXID 1 RXID 1
> 

TXID **_must not_** be equal to RXID !!

pylessard, 2023-02-24:
Yes, IDs seems to conflict.
You can share can log and even isotp logs by enabling the python logger called "isotp".

I think you might be wrongly interpreting what you are seeing

feuerball, 2023-02-25:
> > I got two ecus:
> > **ECU 1** VX1161 Channel 1 TXID 1 RXID 1
> > **ECU2** VX1161 Channel 2 TXID 1 RXID 1
> 
> TXID **_must not_** be equal to RXID !!

They are not. 
I got TXID 1 twice and RXID 1 twice, 
but TXID 1 != RXID 1

pylessard, 2023-02-25:
1=1, pretty sure. :)

feuerball, 2023-02-25:
No.

I can Fill in the actual values if you want.
It is the same ecu Equipped twice. One on chan 1, the Other one on chan 2

pylessard, 2023-02-25:
Hi,
You are hard to follow. Isotp has a requirement txid!=rxid on the same CAN physical line. That's it.

Also, I've asked for can log few times. If you want help, please share a log. I can't base my analysis on your interpretation of the situation (because it looks wrong).

Cheers

feuerball, 2023-02-25:
Hey,

I will try to provide logs, but I am on business trip right now, therefore I can only explain.
Let me try to be more clear :)

I got the following ecu twice.

ECU name : charlie
RXID = 1234
TXID = 5678

I got a VX1161 with multiple CAN Ports
charlie_1 is attached to can 1
charlie_2 is attached to can 2

If I create both busses in parallel and try to do parallel diag requests now, everything is messed up (I think because the bus from charlie_1 gets the response from charlie_2, which does not match to the expected response from charlie_1 obviously).

If I open the first bus first, do the requests for charlie_1, close the bus, open the next bus, do the requests for charlie_2, everything works fine.

Sorry for the mess.

pylessard, 2023-02-25:
Ah. I prefer thoses IDs :)
That sound like something at a lower level. The CAN log will help. IsoTP log as well (enabling "isotp" logger in debug mode)

Check that you have no driver level loopbacks. Vector have this funny tool where you see all your devices and assigns them to application. Go see if there's a virtual bus there looping back on a physical bus.

You can test pythoncan directly and see if the same happen. Then try sending a message with a command line tool to see if the issue is in python-can. If all cases cause issue, it's the driver

pylessard, 2023-02-25:
Stupid question. Do you have a physical loopback in the wires? Like a custom made adapter?

feuerball, 2023-02-25:
Was a bit confusing, sorry for that :D

I will try to contact my colleagues, so they might provide me some logs. 
If I got time this week, I will also try your example.

Hopefully not, otherwise our cars won't drive anymore 🤣

pylessard, 2023-08-13:
Should we close that issue?

closed 2023-10-10 by pylessard