pylessard/python-can-isotp #86: After receivie NRC0x78, isotp cannot receive the follow-up messages.

EricL752, 2023-05-06:
According to the UDS, after receive the NRC0x78, the ECU will send the follow-up data before P2*-timeout. But I find the package can just receive the negative response with 78NRC.

EricL752, 2023-05-06:
Problem is still exist. But solution has been found

pylessard, 2023-05-06:
Maybe you can gives details? Nrc 78 is handled by this module

EricL752, 2023-05-15:
you may misunderstand my description. I found, somtimes the response message could be lost after receiving the NRC0x78, but afterwards I found this issue also occurred without NRC0x78, I'm trying to check my code. But I have a question: do you meet the samilar problem like my issue, when isotp-pkg is using very fast(a request pro 10ms) in a long time(more than one day)

EricL752, 2023-05-15:
Also question: Will NRC0x78 message be received by isotp-pkg? I used CAPL-language with oesk-tp.dll by VECTOR, it will take the  out the NRC78 msg, and just get the following-up msg.

pylessard, 2023-05-15:
isotp receives everything it is supposed to. it has no knowledge of what an NRC is, becaus eit is a UDS paradigm.
I am unable to tell whether I have seen your issue or not because I do not understand what your issue is.
No offenses, but in order to get a quality answer, you need to ask a quality question.

If you want help, describe what you did, show your config, even better: your code.  Show the symptoms, that includes error messages and stack trace. You can enable logging and show the logs.  A raw log of the can layer is also useful.  Tells us what you expect to see and what you think is the problem. Talk about your setup, are you running the user space transport layer or the Linux kernel module?

EricL752, 2023-05-16:
I'm too excited because of my first release an issue. And so sorry about my vague description. I can send you my code after I add the commenting, now I can add some config info:

1. this is the parameter about isotp-pkg:
[image]

2. Action is sending the cyclic UDS requests. In order to find my issue, I print the timestamp and the received 0x22 service's DID results, when I get the tuple(1, 0), it means I don't receive the response.
[image]

3. please take the ASC format log if you need, according the screenshot, the response for DID 0xC004 has been received(in CANOE). it's different with my sight in pycharm
[image]

4. I used VECTOR HARDWARE in my WINDOWS PC, about some questions you asked, I'm not sure, but I can send my code as below:

import datetime
import logging
import time
import sys
import os

from PyQt5 import QtWidgets
from PyQt5.QtCore import QTimer, Qt
from can.interface import Bus
from isotp import Address, AddressingMode, CanStack, TargetAddressType

class SamplePycan:
    interface = "vector"
    app_name = "Canoe"
    channel = 0
    can_filters = [{'can_id': 0x735, 'can_mask': 0x7FF, 'extended': False},
                   {'can_id': 0x73d, 'can_mask': 0x7FF, 'extended': False}]
    receive_own_messages = True
    bitrate = 500000
    data_bitrate = 2000000
    fd = True
    sam_abr = 1
    sjw_abr = 8
    tseg1_abr = 31
    tseg2_abr = 8
    sjw_dbr = 2
    tseg1_dbr = 7
    tseg2_dbr = 2

class SampleTP:
    stmin = 0
    blocksize = 8
    squash_stmin_requirement = False
    rx_flowcontrol_timeout = 75
    rx_consecutive_frame_timeout = 150
    tx_padding = 0xAA
    wftmax = 0
    tx_data_length = 8
    tx_data_min_length = 8
    max_frame_size = 4095
    can_fd = True
    bitrate_switch = True
    default_target_address_type = TargetAddressType.Physical
    rate_limit_max_bitrate = 10000000
    rate_limit_window_size = 0.2
    rate_limit_enable = True
    listen_mode = False

class Sample(object):
    P2client = 0.1
    P2star = 5.0
    res_id = 0x73d
    req_id = 0x735

    def __init__(self):
        super().__init__()
        can_param = SamplePycan()
        self.can_param = dict((name, getattr(can_param, name)) for name in dir(can_param) if not name.startswith('__'))
        self.bus = Bus(**self.can_param)
        tp_param = SampleTP()
        self.tp_param = dict((name, getattr(tp_param, name)) for name in dir(tp_param) if not name.startswith('__'))
        self.addr_phy = Address(AddressingMode.Normal_11bits, rxid=self.res_id, txid=self.req_id)
        self.stack = CanStack(self.bus, address=self.addr_phy, error_handler=self.my_error_handler,
                              params=self.tp_param)

    @staticmethod
    def my_error_handler(error):
        logging.warning('IsoTp error happened : %s - %s' % (error.__class__.__name__, str(error)))

    def tp_send(self, data: bytearray):
        self.stack.send(data)
        time_to_p2c = time.process_time() + self.P2client
        while time.process_time() - time_to_p2c < 0:
            self.stack.process()
            if self.stack.available():
                payload = list(self.stack.recv())
                return payload
        return [0, ]

class Sender:
    def __init__(self, msec, func):
        self.timer = QTimer()
        self.func = func
        self._timer(msec)

    def _timer(self, msec=None):
        if msec:
            self.timer.setTimerType(Qt.PreciseTimer)
            self.timer.timeout.connect(self.func)
            self.timer.start(msec)
        else:
            raise ValueError

class MessageRecv(Sample):
    def __init__(self):
        super().__init__()
        self.st = Sender(100, self.send_msg_ch0)

    def send_msg_ch0(self):
        r000 = self.tp_send(bytearray(b"\x22\xc0\x00"))
        r001 = self.tp_send(bytearray(b"\x22\xc0\x01"))
        r002 = self.tp_send(bytearray(b"\x22\xc0\x02"))
        r003 = self.tp_send(bytearray(b"\x22\xc0\x03"))
        r004 = self.tp_send(bytearray(b"\x22\xc0\x04"))
        r005 = self.tp_send(bytearray(b"\x22\xc0\x05"))
        r006 = self.tp_send(bytearray(b"\x22\xc0\x06"))
        if (r000[0] == 0) or (r001[0] == 0) or (r002[0] == 0) or (r003[0] == 0) or (r004[0] == 0) or \
                (r005[0] == 0) or (r006[0] == 0):
            print(time.time(), datetime.datetime.now(), (len(r000), r000[0]), (len(r001), r001[0]),
                  (len(r002), r002[0]), (len(r003), r003[0]), (len(r004), r004[0]), (len(r005), r005[0]),
                  (len(r006), r006[0]))

if __name__ == '__main__':
    app = QtWidgets.QApplication(sys.argv)
    mr = MessageRecv()
    app.exec_()
    # time.sleep(10)
    # os._exit(0)

EricL752, 2023-05-17:
thanks for your advice, and I found my ECU has some issue, it caused, the CAN bus receive the empty can frame.
I think, this ECU issue affects indirectly the code running. After the ECU issue fix, I will have a retry with your thread-safe advice.

EricL752, 2023-05-17:
I retried. And this issue didn't reproduce, as the below screenshot, ECU responsed us in around 100ms, it's illegal for P2Service-timeout according ISO-15765, right? The isotp pkg just handle the response within 50ms, right? 
[image]
If my thought is right, the issue can be closed, if you also think so.
thank you very much once again.

pylessard, 2023-05-17:
It's not illegal. That depends on your setup.
First P2 timeout is a UDS concept. ISOTP does not know about it. ISOTP timeouts are differents and defaults values are 1000ms.

For P2 timeout, 100ms is not illegal, depends on your server
Be careful though. With UDS latest version, the server will tell the client what are its timeout value when responding the DiagnosticSessionControl service. and the client from the udsoncan package will automatically switch to these timeouts value when received. you can disable that feature by setting  the use_server_timing parameter to False.

Cheers

EricL752, 2023-05-19:
Understood. I want to downgrade my ECU Version to elder version. and reproduce this issue, becuase I suspect, maybe errorframes are received at that time, and the issue occurs. Anyhow, Cheers first. :)

EricL752, 2023-05-19:
Retried and not reproduce. close the issue and thanks for your support 👍

closed 2023-05-19 by EricL752