pylessard/python-can-isotp #87: still absent support of CAN bus timeouts

allnet12, 2023-06-18:
Hello,

I found that this commit was not included in the recent release 1.8:
"Enhanced compatibility with python-can
mikisama committed on Apr 29, 2022"

https://github.com/pylessard/python-can-isotp/commit/c0e4b7f966c5945b8d484c744aeb030c9cac7d85

Was it missed or not included for some reason?

Thanks,
Alexander

pylessard, 2023-08-12:
hey, sorry for not answering. I did not notice this issue. All on me

you are right, it has not been included. Looking at the commit history, I realize that I should make 1.9 release, there's a bunch of changes that has been added since v1.8. Will publish 1.9 really soon

pylessard, 2023-08-12:
There you go my good sir. V1.9.0 is released including the commit you pointed out.
Thank you!

closed 2023-08-12 by pylessard