pylessard/python-can-isotp #88: Rate limiter still works strange

pompushko, 2023-08-09:
I have factory OBD flasher which works very well and fast.

I wrote own flasher, which works same as factory. But with issues of timeout after some time after running transfer_data:
```
udsoncan.exceptions.TimeoutException: Did not receive response in time. Global request timeout time has expired (timeout=50.000 sec)
```
Here is my settings, which load flash with udsoncan.client.transfer_data:
```
isotp_params = {
   'stmin' : 32,                          # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
   'blocksize' : 8,                       # Request the sender to send 8 consecutives frames before sending a new flow control message
   'wftmax' : 0,                          # Number of wait frame allowed before triggering an error
   'tx_data_length' : 8,                  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
   'tx_data_min_length' : None,           # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
   #'tx_padding' : 0,                      # Will pad all transmitted CAN messages with byte 0x00.
   'rx_flowcontrol_timeout' : 2000,       # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
   'rx_consecutive_frame_timeout' : 2000, # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
   'squash_stmin_requirement' : False,    # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
   'max_frame_size' : 4095,               # Limit the size of receive frame.
   'rate_limit_enable' : True,
   'rate_limit_max_bitrate' : 1000,
   'rate_limit_window_size': 1
}
```

But it takes hours...

Also, if I disable rate_limit_enable - server device can stop answer after first part of transfer_data. If I enable it - I can load few parts of flash...

Could you please help me, thank you.

pylessard, 2023-08-09:
Do you have a raw log?
What's the size of the transferred binary?
what's your bitrate?

1000bps is very very low.

pompushko, 2023-08-09:
Sure!

```
can2  18DA20F1   [8]  10 0B 34 03 44 A0 01 C0
  can2  18DAF120   [3]  30 00 00
  can2  18DA20F1   [6]  21 00 00 04 40 00
  can2  18DAF120   [4]  03 7F 34 78
  can2  18DAF120   [5]  04 74 20 0F E2
  can2  18DA20F1   [8]  10 82 36 01 DF 3F FE 5F
  can2  18DAF120   [3]  30 00 00
  can2  18DA20F1   [8]  21 FF FF FF 55 FF 5F C3
  can2  18DA20F1   [8]  22 00 6F 34 FE 7F 4F 35
  can2  18DA20F1   [8]  23 FE 7F 0F 37 FE 7F FE
  can2  18DA20F1   [8]  24 FF FF FF 1F 87 FC 7F
  can2  18DA20F1   [8]  25 62 79 49 D8 F5 FF FF
  can2  18DA20F1   [8]  26 FF FF FE FF FE FF FE
  can2  18DA20F1   [8]  27 FF FE FF FF F0 FF FF
  can2  18DA20F1   [8]  28 FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  29 FC FF FF FF FF FF FF
  can2  18DA20F1   [8]  2A FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  2B FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  2C FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  2D FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  2E FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  2F FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  20 FF FF FF FF FF FF FF
  can2  18DA20F1   [8]  21 FF FF FF FF FF FF FF
  can2  18DA20F1   [6]  22 FF FF FF FF FF
  can2  18DAF120   [4]  03 7F 36 78
  can2  18DAF120   [3]  02 76 01
  can2  18DBFEF1   [3]  02 3E 80
  can2  18DA20F1   [8]  10 82 36 02 10 54 BA FE
  can2  18DAF120   [3]  30 00 00
  can2  18DA20F1   [8]  21 10 54 BA FE 10 54 BA
  can2  18DA20F1   [8]  22 FE 10 54 BA FE 10 54
  can2  18DA20F1   [8]  23 BA FE 10 54 BA FE 10
  can2  18DA20F1   [8]  24 54 BA FE 10 54 BA FE
  can2  18DA20F1   [8]  25 11 54 BA FE 10 54 BA
  can2  18DA20F1   [8]  26 FE 10 54 BA FE 10 54
  can2  18DA20F1   [8]  27 BA FE 10 54 BA FE 10
  can2  18DA20F1   [8]  28 54 BA FE 10 54 BA FE
  can2  18DA20F1   [8]  29 10 54 BA FE 13 54 BA
  can2  18DA20F1   [8]  2A FE 10 54 BA FE 10 54
  can2  18DA20F1   [8]  2B BA FE 10 54 BA FE 10
  can2  18DA20F1   [8]  2C 54 BA FE 10 54 BA FE
  can2  18DA20F1   [8]  2D 10 54 BA FE 10 54 BA
  can2  18DA20F1   [8]  2E FE 17 54 BA FE 10 54
  can2  18DA20F1   [8]  2F BA FE 10 54 BA FE 10
  can2  18DA20F1   [8]  20 54 BA FE 10 54 BA FE
  can2  18DA20F1   [8]  21 10 54 BA FE 10 54 BA
  can2  18DA20F1   [6]  22 FE 10 54 BA FE
```

here is candump when I trying to transfer data.
even I got problems when I load 128b chunk...

500kb\s my interface. 

even I make dump of canbus when OEM factory flasher load flash.

pylessard, 2023-08-10:
Would be helpful if you had timestamps in your dump.

if the bus is 500kbps, why do you limit it to 1000bps with the rate limiter?

pompushko, 2023-08-10:
> Would be helpful if you had timestamps in your dump.
> 
> if the bus is 500kbps, why do you limit it to 1000bps with the rate limiter?

Let me try add timestamps...

Which limiter should be?

pylessard, 2023-08-10:
Sorry I don't understand your question.  You enable a rate limiter on the bus with a value of 1000bps and you say that transfer is slow.. Seems related to me.

pompushko, 2023-08-10:
Im sorry. Misunderstanting.

Main problem , is a timeout. I tried to use rate limiter to avoid of timeout problem. (I thought my device think about throttling)

pylessard, 2023-08-10:
Okay.  The more detail you give, the more helpful I can be.

When I look at the logs, I see 2 transfer of data and the second one gets no response.  The sender side seems correct, it's just that the device never respond, so the timeout seems legit.

pompushko, 2023-08-10:
Yes, You right... First data transfer is okey. Second is not.
Sometimes, I can load 3-4 chunks of memory. And after loadting 5 chunk - no response

For example, my code can load one or two chunk of memory to device. After that, I dont get Response like **02 76 XX** ...

pylessard, 2023-08-10:
Ok, I think there is no issue on the ISOTP layer.   Either your device crashes/reset or you are not following the correct flash procedure for your device and it expects something else than a TransferData message.

Hard to tell more

pylessard, 2023-08-10:
This issue has been handled through another communication channel. 
Issue was that device needed a delay between can messages and given stmin was 0 (no limit).  Forcing a slower transmit stmin solved the issue. 
@pompushko : You are free to add information if you want.

closed 2023-08-10 by pylessard

hartkopp, 2023-08-12:
> This issue has been handled through another communication channel. Issue was that device needed a delay between can messages and given stmin was 0 (no limit). Forcing a slower transmit stmin solved the issue. @pompushko : You are free to add information if you want.

Finding such issue with a CAN logfile without timestamps is challenging ;-)