pylessard/python-can-isotp #89: Custom counter for consecutive frame

pompushko, 2023-08-10:
Hello :)

Many OEM manufacturers have own small crypto algoritm during eg. TransferData (0x36) and they use for that service special pattern of sequence\index number. Let me try to explain.

Here is a can dump from OEM software:
`
   30.1417 1  18DA20F1x         Tx   d 8 1F E2 36 01 DF 3F FE 5F
   30.1417 1  18DAF120x         Rx   d 3 30 00 00
   30.1441 1  18DA20F1x         Tx   d 8 21 FF FF FF 55 FF 5F C3
   30.1491 1  18DA20F1x         Tx   d 8 25 10 54 BA FE 10 54 BA
   30.1551 1  18DA20F1x         Tx   d 8 29 EF AB 45 01 EF AB 45
   30.1601 1  18DA20F1x         Tx   d 8 2D CD 23 67 89 CD 23 67
   30.1661 1  18DA20F1x         Tx   d 8 21 32 DC 98 76 32 DC 98
   30.1721 1  18DA20F1x         Tx   d 8 25 FF FF FF FF FF FF FF
   30.1771 1  18DA20F1x         Tx   d 8 29 FF FF FF FF FF FF FF
   30.1831 1  18DA20F1x         Tx   d 8 2D FF FF FF FF FF FF FF
`
As you could see, in that case, during TransferData (0x36) it use only 1-5-9-D counter index only.
And here is a log file from OEM software.

`
<line> tx: CANCH29BIT - (NP) 18DA20F1 - 36 01 DF 3F FE 5F FF FF FF 55 FF 5F C3 00 6F 34 FE 7F 4F 35 FE 7F 0F 37 FE 7F FE FF FF FF 1F 87 FC 7F 62 79 53 D8 F5 FF FF FF FF FE FF FE FF FE FF FE FF FF F0 FF FF FF FF FF FF FF FF FF FC FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 11 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 13 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 17 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE 10 54 BA FE EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 11 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 31 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 71 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 EF AB 45 01 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 99 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 B9 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 F9 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 CD 23 67 89 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 33 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 37 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 3F DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 32 DC 98 76 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 9F 00 FE 7F FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF ...</line>
`

pylessard, 2023-08-10:
Can probably easily done in the python implementation, more complex in the kernel driver.
This is non-standard isotp and I have my doubts that it is used as a crypto mechanism.  

You can try to modify this line. https://github.com/pylessard/python-can-isotp/blob/master/isotp/protocol.py#L670

If I were you, I would pass a callback tom compute the next counter which default to the normal behavior.
something like : 

```python
def next_counter(last_counter):
    return (last_counter + 1 ) & 0x0F
```
And in your case 

```python
def next_counter(last_counter):
    return (last_counter + 4 ) & 0x0F
```

pompushko, 2023-08-10:
Great! Thank you!

This crypto mechanism is simple: 

1. We have bigger binary file than needed to be transferred. Like obfuscation
2. We send that binary data as is via eg. TransferData (0x36)
3. But send only 1-5-9-D counters.

pylessard, 2023-08-10:
I would not put such change into this repo as it is not a behavior stated by iso-15765. You can do it in a fork if you want.

btw, as discussed outside of github, I still think you can transceiver is not picking all the can frames.

pompushko, 2023-08-10:
Well.. I used 2 different canbus sniffers from 2 sources. And thats as it works ;(
Also, thats why TransferData faster than in my case.

hartkopp, 2023-08-12:
OMG! Who's designing such bad "Security by Obscurity" ?? :-D

I'm not sure whether incrementing the SN counter by 4 (instead of 1) really covers the entire story.

In the PDU example the 'skipped' content would be 3 x 7 = 21 byte. But the PDU looks different:
```
   30.1417 1  18DA20F1x         Tx   d 8 1F E2 36 01 DF 3F FE 5F
   30.1417 1  18DAF120x         Rx   d 3 30 00 00
   30.1441 1  18DA20F1x         Tx   d 8 21 FF FF FF 55 FF 5F C3
   30.1491 1  18DA20F1x         Tx   d 8 25 10 54 BA FE 10 54 BA
   30.1551 1  18DA20F1x         Tx   d 8 29 EF AB 45 01 EF AB 45
```
First Frame data
`36 01 DF 3F FE 5F` 

Consecutive Frame data
`FF FF FF 55 FF 5F C3` 

Filler data - 0x75 (117 dec) byte ??
`00 6F 34 FE 7F 4F 35 FE 7F 0F 37 FE 7F FE FF FF`
`FF 1F 87 FC 7F 62 79 53 D8 F5 FF FF FF FF FE FF`
`FE FF FE FF FE FF FF F0 FF FF FF FF FF FF FF FF`
`FF FC FF FF FF FF FF FF FF FF FF FF FF FF FF FF`
`FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF`
`FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF`
`FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF`
`FF FF FF FF FF`
(maybe more filler bytes as the following pattern repeats several times)

Next consecutive frame data
`10 54 BA FE 10 54 BA`

Next filler ...
`FE 10 54 BA FE 10 ...`

Do you have a real number of fill data between the 'real' consecutive frame content?

pylessard, 2023-08-12:
Nobody does. These conversation are scattered across github, emails and linkedin. The story isn't full. The author confirmed that his socketcan setup was dropping frames. The comm is using normal counter.

I'm closing this :)

closed 2023-08-12 by pylessard