pylessard/python-can-isotp #90: Replcaing the bytearray data handling with Generator

geynis, 2023-08-27:
As CAN-FD supports up to 4GB of data, the code should not convert the data to bytearray. Instead the data should be handled as an iterator of bytes to let the user use a byte generator.

pylessard, 2023-08-27:
CAN-FD supports frames up to 64 bytes.
You probably are referring to the ISO-TP frame size as defined since the 2016 version where data length is 32 bits (instead of 12 bits as before).

I understand the request. Do you have a real use-case for it? 
IF this gets implemented, it would need to be a pair of generator and max length since the payload size needs to be known at the beginning of the transmission

geynis, 2023-08-28:
Hi, yes, I'm referring to the max ISO-TP length for CAN-FD.
Yes it can be a a pair with max length or it can be a single object that
implements __len__ .

geynis, 2023-08-28:
BTW, if you just remove any casting to bytearray the user could provide a
costume object that will generate the data on the fly.

pylessard, 2023-08-28:
Can I know what kind of use case requires that?

geynis, 2023-08-28:
Cyber Security Testing

pylessard, 2023-08-28:
Ok, this will go in the todo list.   You can propose a PR (with unit test) if you want.

geynis, 2023-08-28:
Yes, Thanks!

pylessard, 2023-12-30:
Will be available in v2.x

closed 2023-12-30 by pylessard

geynis, 2024-01-01:
hi @pylessard , thanks for the implementation. I don't think you need to implement it as a tuple, generators can return length with the _len_ function. e.g. _range_ is a generator and has a length without generating all the elements.

pylessard, 2024-01-02:
Hi @geynis , I know generators can have a length, but they are not required to. So either I ask the user to provide a custom generator that implements the __len__ function or I ask for the length explicitely.  I find the API for a tuple more simple, even if it requires few additional characters (a comma and paranthesis). 

In my experience, an explicit API avoid confusion and weird bugs. If a user implements the __len__ function, the user may think that it should return the remaining number of element instead of the total, which would be an issue. It's just a dumb example.

Cheers

PR #106 (2023-12-30): Support generator for send
Fixes #90