pylessard/python-can-isotp #101: How to set stmin when use transfer_data()

trandangkhoadev, 2023-11-29:
I'm using Transport layer and PythonIsoTpConnection but when I call transfer_data() the stmin will respect the stmin requirement of receiver. 
In case stmin from receiver as 0. The transfer_data() will send the cantp frame to fast such as stmin= 0.2ms. So how can I increase the delay between 2 frames in this case.

trandangkhoadev, 2023-11-29:
When I use ver 2.x and set stmin = 0.001 but the actual stmin is 16ms. So how can I have the stmin 0.5 ms ? Because my hardware only work with stmin = 0.5ms

pylessard, 2023-11-29:
That's progress!
I'm actively working into improving timings.  It's hard to do in a fully portable fashion in pure-python.
Can you try to move you udsoncan repo to the `support-isotp-v2` branch?    Hopefully, you will get better timings.

Note that I still have some work to do, like enabling the use of the CAN notifier which seems to provide betterperformance as pointed out in this issue: https://github.com/pylessard/python-udsoncan/issues/169

trandangkhoadev, 2023-11-30:
So that means, if we only set override_receiver_stmin to 0, it actually results in a delay of 0.2ms, and if we set override_receiver_stmin to 0.0001, it actually causes a delay of 16ms. Is there no other way to achieve a delay in the range of 0.5ms to 16ms?

pylessard, 2023-12-21:
following up on this?
I may close this issue if no answer. cheers

closed 2023-12-29 by pylessard