pylessard/python-can-isotp #110: TypeError: dataclass() got an unexpected keyword argument 'slots'

matthuisman, 2024-01-09:
Python 3.7 erroring with the below message

`    import isotp
.envs\py37\venv\lib\site-packages\isotp\__init__.py:6: in <module>
    from isotp.protocol import TransportLayerLogic, TransportLayer, CanStack, NotifierBasedCanStack
.envs\py37\venv\lib\site-packages\isotp\protocol.py:300: in <module>
    class TransportLayerLogic:
.envs\py37\venv\lib\site-packages\isotp\protocol.py:303: in TransportLayerLogic
    @dataclass(slots=True, init=False)
E   TypeError: dataclass() got an unexpected keyword argument 'slots'`

pylessard, 2024-01-09:
You didn't loose time here to report :)
Will fix right away

matthuisman, 2024-01-09:
yeh, it broke a lot of my works CI :) so we caught it quick
Thanks

pylessard, 2024-01-09:
Last release is a major version number change. I broke many interface

pylessard, 2024-01-09:
v2.0.1 is release and published.  Should worked.

closed 2024-01-09 by pylessard

matthuisman, 2024-01-09:
thank you for the fast turn around. much appreciated

commit 78f1ce34 (2024-01-09): Fixed broken compatibility with python 3.7 (#111)

Fixes #110