pylessard/python-can-isotp #113: Performance on Windows

pgreenland, 2024-01-22:
Hey,

Firstly thanks for an awesome library!

I've been using it on Linux and now Windows as part of an ECU firmware updater. So been using the pure Python implementation.

It's been working well on both, however performance on Windows has been terrible compared to Linux.

As an example on Linux the main file transfer takes 10s, on Windows it takes 2mins.

I believe I've narrowed it down to the Timer implementation, which calls `time.monotonic_ns`. The resolution of that Timer on Windows doesn't appear to be very good.

Based on the discussion of `time.monotonic` here https://bugs.python.org/issue44328.

Tacking the following into my code resolves the issue:

```
if platform.system() == "Windows":
    # On windows patch time.monotonic_ns for isotp library.
    # It seems time.monotonic uses a timer which has a resolution of 16ms
    # See: https://bugs.python.org/issue44328
    # Instead we'll override it with our own version
    time.monotonic_ns = lambda: time.perf_counter() * 1000000000.0
```

Bringing the update time down to around 13-15s....not quite as good as Linux....but it's close enough. Watching the CAN traffic on another machine the inter-frame delay drops from 10-12ms to 1ms as requested by the remote device.

Thought it may be beneficial to others to fix the problem at source.

Thanks,

Phil

pylessard, 2024-01-22:
oh wow.
With isotp v2, I've spent a lot of effort to redesign the module the remove all timing weakness. This big one went unnoticed.
I will fix this very very soon.
Thank you

pgreenland, 2024-01-22:
Wow that was rapid, thanks!

I'll give it a go on both platforms later today and report back. Im wondering if those extra timers you tweaked may be responsible for my additional few seconds  😄 .

Regards,

Phil

pylessard, 2024-01-22:
Possibly, if you use the rate limiter feature, it could be.
The other one is only a protection to avoid bloating a core, so I doubt it is.

pgreenland, 2024-01-22:
Installed on Windows and Linux (removing my hack from Windows).

Working great in both. Windows is still a tiny bit slower, but eh....its Windows you get what you pay for :-P

Thanks for sorting it so quickly.

Phil

pylessard, 2024-01-22:
Awesome. Expect v2.0.3 very shortly

pylessard, 2024-01-23:
Fixed in #114 
Released into v2.0.3

Thank you again, that was probably the last missing piece of the timing puzzle.
Cheers

closed 2024-01-23 by pylessard