pylessard/python-can-isotp #117: timed out isotp_socket.recv()

pompushko, 2024-02-09:
Hello

I upgraded isotp up to the last version and my old app stop works :D

```
import isotp
from uds import routines
from loggers.logger_app import logger

def start(CAN_INTERFACE, ECU_NAME, UDS_ECU_ADDRESS, UDS_TARGET_ADDRESS):
    isotp_socket = create_isotp_socket(CAN_INTERFACE, UDS_ECU_ADDRESS, UDS_TARGET_ADDRESS)
    while True:
        request = isotp_socket.recv()
        if request is not None:
            log_request(ECU_NAME, UDS_ECU_ADDRESS, UDS_TARGET_ADDRESS, request)
            if len(request) >= 1:
                response = routines.process_service_request(ECU_NAME, request)
                if response is not None:
                    log_response(UDS_ECU_ADDRESS, response)
                    isotp_socket.send(response)

def create_isotp_socket(CAN_INTERFACE, receiver_address, target_address):
    socket = isotp.socket(timeout=1)
    socket.bind(CAN_INTERFACE, isotp.Address(isotp.AddressingMode.Normal_29bits, rxid=receiver_address, txid=target_address))
    return socket
```

```
Exception in thread Thread-1 (start):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
Exception in thread Thread-2 (start):
Traceback (most recent call last):
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
  File "/home/pi/ecu-simulator/uds/listener.py", line 8, in start
    request = isotp_socket.recv()
              ^^^^^^^^^^^^^^^^^^^
  File "/home/pi/.local/lib/python3.11/site-packages/isotp/tpsock/__init__.py", line 113, in recv
    return self._socket.recv(bufsize, flags)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TimeoutError: timed out
```

Maybe I have some problems with my thread creation? Thank you

pylessard, 2024-02-09:
No, check the release note. Recv throws timeout instead of returning None with v2.x

pylessard, 2024-02-09:
[image]

pompushko, 2024-02-09:
Great, could you please help me, what to change, to get it works :D?
Unfortunately, I`m not well in Python yet...

pylessard, 2024-02-09:
Comes from #46

pylessard, 2024-02-09:
```
try:
...
except socket Timeout error: 
    pass
```

pompushko, 2024-02-10:
> ```
> try:
> ...
> except socket.TimeoutError: 
>     pass
> ```

Unfortunately, cant get it works:
```
    except isotp_socket.TimeoutError:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'socket' object has no attribute 'TimeoutError'
```

pylessard, 2024-02-11:
``TimeoutError`` is defined in the python ``socket`` module.

```
import socket

try:
    ...
except socket.TimeoutError:
    pass
```

closed 2024-02-11 by pylessard