pylessard/python-can-isotp #121: Able to RX/TX ISO-TP frames using Linux can-utils package but not via Python

HAblueHA, 2024-03-06:
I was able to use your great work with several cars but I am now playing with an IVI and a cluster that are capricious...

I am able able to send and receive ISO-TP packets via **isotpsend** and **isotprecv** from can-utils without any problem
However, passing through Python using such a script as below, TX would be successful but never RX : 
```
import isotp
tp = isotp.socket(5)
tp.set_fc_opts(stmin=5, bs=10)
tp.bind('can0', isotp.Address(rxid=0x763, txid=0x743))

tp.send(bytearray([0x10, 0x03]))
print(tp.recv())
```

Linux kernel : 5.10.0-28-amd64
I am working via a VM (as always, even with my previous targets)

Is it the IVI that is too strict in terms of timings or what ? If you have any suggestion...
Thanks in advance for your feedback

pylessard, 2024-03-06:
You need a can log. Impossible to tell what's wrong without a log.

HAblueHA, 2024-03-07:
While generating the log using isotpdump, I realized it was a simple padding problem.
Thank you for your answer

closed 2024-03-07 by HAblueHA

hartkopp, 2024-03-07:
Good to read that someone is using that tool - with success :-D