pylessard/python-can-isotp #123: AttributeError: 'CanMessage' object has no attribute 'channel'

daekp, 2024-04-25:
I am attempting to use Vector for CANTP communication in a Windows. When I try to send data, I encounter an AttributeError in Python. Do you know what might be causing this?

`
class TEST:
    def __init__(self):
        self.transport_layer = None
        self.thread_list = []
        self.config = json.load(open('config.json'))
        self.can_bus = VectorBus(channel=self.config['channel'],
                                 interface='vector',
                                 app_name='python-can',
                                 serial=self.config['vector_serial'],
                                 bitrate=self.config['bitrate'],
                                 fd=self.config['fd'])

    def setup_isotp_transport(self):
        address = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x111, rxid=0x222)
        transport_layer = isotp.TransportLayer(address=address,
                                               txfn=self.can_bus.send,
                                               rxfn=self.can_bus.recv)
        return transport_layer

    def send_data(self, data: list):
        can_id = 0x111
        transport_layer = self.setup_isotp_transport()
        transport_layer.start()
        large_data = bytearray(data)
        transport_layer.send(large_data, target_address_type=0)

    def transmitter(self):
        sequence = {
            "ExtendedSessionControl": [0x02, 0x10, 0x03, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA],
            "Data": [0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]
        }

        for key in sequence:
            self.send_data(sequence[key])

    def receiver(self):
        while True:
            message = self.can_bus.recv(timeout=.0005)

    def run_threads(self):
        transmitter_thread = threading.Thread(target=self.transmitter)
        receiver_thread = threading.Thread(target=self.receiver)

        self.thread_list.append(transmitter_thread)
        self.thread_list.append(receiver_thread)

        transmitter_thread.start()
        receiver_thread.start()

if __name__ == '__main__':
    re = TEST()
    re.run_threads()
`

pylessard, 2024-04-25:
Hi,
Your integration with python-can is wrong. IT has multiple problem.
First, the txfn,rxfn parameter of the TransportLayer have a signature requirement that is not directly compatible with PythonCan.
Also, the TransportLayer makes its own internal thread. I doubt you need to create more.

For an integration with PythonCan, you can use the ``CanStack`` object which is nothing more than a ``TransportLayer`` preconfigured with correct txfn/rxfn.

See https://can-isotp.readthedocs.io/en/latest/isotp/implementation.html#isotp.NotifierBasedCanStack

pylessard, 2024-05-27:
Any update? Should we close this issue?