pylessard/python-can-isotp #124: The Listener can not log can message correctly

jeffery96, 2024-05-17:
Hi pylessard,

I'm trying to log can message by adding can.ASCWrite('./test.asc') to listener list, but it just created the test.asc file without any record in the file while indeed there are can messages transmitting(the can.Printer can print the message correctly).

I try pass a IO type to can.ASCWrite(), it did not work. Then I add  IO.close() at the end of code, it work.
[image]
[image]

Here is my code:

```
import time

import can
from can.interfaces.pcan import PcanBus
from udsoncan.connections import PythonIsoTpConnection
from udsoncan.client import Client
import udsoncan.configs
import isotp

# Refer to isotp documentation for full details about parameters
isotp_params = {
    'stmin': 32,
    # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
    'blocksize': 8,  # Request the sender to send 8 consecutives frames before sending a new flow control message
    'wftmax': 0,  # Number of wait frame allowed before triggering an error
    'tx_data_length': 8,  # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
    # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
    'tx_data_min_length': None,
    'tx_padding': 0x55,  # Will pad all transmitted CAN messages with byte 0x00.
    'rx_flowcontrol_timeout': 1000,  # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
    'rx_consecutive_frame_timeout': 1000,
    # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
    'override_receiver_stmin': None,      # When sending, respect the stmin requirement of the receiver. If set to True, go as fast as possible.
    'max_frame_size': 4095,  # Limit the size of receive frame.
    'can_fd': False,  # Does not set the can_fd flag on the output CAN messages
    'bitrate_switch': False,  # Does not set the bitrate_switch flag on the output CAN messages
    'rate_limit_enable': False,  # Disable the rate limiter
    'rate_limit_max_bitrate': 1000000,
    # Ignored when rate_limit_enable=False. Sets the max bitrate when rate_limit_enable=True
    'rate_limit_window_size': 0.2,
    # Ignored when rate_limit_enable=False. Sets the averaging window size for bitrate calculation when rate_limit_enable=True
    'listen_mode': False,  # Does not use the listen_mode which prevent transmission.
}

udsoncan.setup_logging()

uds_config = udsoncan.configs.default_client_config.copy()
# uds_config['use_server_timing'] = False
# uds_config['request_timeout'] = None

bus = PcanBus(bitrate=250000)   # Link Layer (CAN protocol)
time.sleep(0.01)                

# file_to_write = open(file='./test.asc', mode='w+', encoding='utf-8')    # this work with add file_to_write.close() at the end.

# notifier = can.Notifier(bus, [can.Printer(), can.ASCWriter(file_to_write)])  # this work with add file_to_write.close() at the end.
notifier = can.Notifier(bus, [can.Printer(), can.ASCWriter('./test.asc')])  # Add a debug listener that print all messages
tp_addr = isotp.Address(isotp.AddressingMode.Normal_11bits, txid=0x123, rxid=0x456)  # Network layer addressing scheme
# stack = isotp.CanStack(bus=bus, address=tp_addr, params=isotp_params)              # isotp v1.x has no notifier support
stack = isotp.NotifierBasedCanStack(bus=bus, notifier=notifier, address=tp_addr, params=isotp_params)  # Network/Transport layer (IsoTP protocol). Register a new listenenr
conn = PythonIsoTpConnection(stack)  # interface between Application and Transport layer

with Client(conn, config=uds_config) as client:  # Application layer (UDS protocol)
    client.change_session(3)
    client.change_session(1)
    # while True:
    #     pass
    # ...

# file_to_write.close()    # this work with add file_to_write.close() at the end.
```
I‘m not sure where I make mistake when I pass StringPathLike to can.ASCWrite().

Thanks.

pylessard, 2024-05-22:
Hmmm, I don't see why this wouldn't work. Is there messages printed on the console? 

You can try enabling debug logging

pylessard, 2024-05-22:
It just mean that the writer is buffered. Closing the file flush the content to the disk.

The correct way of handling that would be to call notifier.stop() it should call stop() on all listeners and the ASCWriter will flush to disk on a stop according to the documentation.

pylessard, 2024-05-22:
Btw, the client and the stack does not stop your notifier to avoid breaking your application if you do something else with it. The stack adds a listener when started and removes it when done. That is all. You must handle your notifer in your app

jeffery96, 2024-05-23:
Hi pylessard,

I've got it how to deal with this problem.

Thanks for your reply.

closed 2024-05-23 by jeffery96