pylessard/python-can-isotp #125: Preformance issue of func "specific_wait_frame"

XXIN0, 2024-05-23:
Hi pylessard,

**softversion**
can-isotp:  2.0.4
computer: win10
python:  3.8.10 32bit

When I  send a uds(service 36) packet complete, it always  wait 20ms-30ms for the program to send the next packet of data. Through testing I found the func `specific_wait_frame` always wait  some time, how to minimize the `specific_wait_frame`  wait time?

[image]
[image]
[image]

pylessard, 2024-05-23:
Hi,
What connection do you use?
Can you share the python logging output with debug logging enabled?

Windows has a thread resolution of 16ms, so you certainly have 2 time slices of wait time.  It really depends on your setup and the underlying library.   Possible that the underlying library is wasting a time slice before passing the data up.

XXIN0, 2024-05-24:
HI,
Hardware：vector VN5640

log info:
check the attachment file for the tp log.
[tp_log.txt](https://github.com/pylessard/python-can-isotp/files/15426651/tp_log.txt)
[image]

XXIN0, 2024-05-24:
I found that I can configure this sleep  timing through the function `set_sleep_timing` to reduce the wait time.

pylessard, 2024-05-25:
You have not shared the connection you used. And I don't see the the startup in the logs.
set_sleep_timing is deprecated and is not supposed to have an effect.

With v2.x, your rxfn is supposed to be blocking and you should call "start" on isotp stack. Check : https://can-isotp.readthedocs.io/en/latest/#v2-x-changes

if changing the sleep_timing has an effect, the code is trying to cope with backward compatibility, probably here : 
https://github.com/pylessard/python-can-isotp/blob/v2.x/isotp/protocol.py#L1609

If your rxfn has a timeout parameter and that parameter is used to do a blocking read, this should not happen. Performance will be much better with blocking read. 

If you show how you setup your stack, I can point where the issue is.

XXIN0, 2024-05-27:
I'm using   `PythonIsoTpConnection` in  `udsoncan.connections ` to connect  the `isotp`.
[image]

start up log.
[image]

It seems that the code here has done blocking processing.
[image]

XXIN0, 2024-05-27:
I upgraded the new udsoncan library and your implementation was updated,this problem is solved.

pylessard, 2024-05-27:
Ah, yes. v1.21+ is required to take advantage of the timing improvements introduced in isotp v2.
What version were you using?

regards

closed 2024-05-27 by pylessard

XXIN0, 2024-05-27:
> Ah, yes. v1.21+ is required to take advantage of the timing improvements introduced in isotp v2. What version were you using?
> 
> regards

v1.13.1, hhh, It's a lot of versions behind the new version