pylessard/python-can-isotp #128: UDS request with BRS enabled and ESI Disables is not working through CanStack request

karthik-bammidi, 2024-07-01:
Hi, 

I'm facing issue when i send Normal_29bits uds request through vector module.  I don;t receive response for the request. After debugging the difference found the difference.

Observed Request on Can Bus:
[image]

Expected Request on Can Bus:
[image]

I have set the Bitrate_switch value to True still i'm not getting response.

````
import can
import isotp
import time

isotp_params = {
 'stmin': 32,                            # Will request the sender to wait 32ms between consecutive frame. 0-127ms or 100-900ns with values from 0xF1-0xF9
 'blocksize': 8,                         # Request the sender to send 8 consecutives frames before sending a new flow control message
 'wftmax': 0,                            # Number of wait frame allowed before triggering an error
 'tx_data_length': 64,                    # Link layer (CAN layer) works with 8 byte payload (CAN 2.0)
 # Minimum length of CAN messages. When different from None, messages are padded to meet this length. Works with CAN 2.0 and CAN FD.
 'tx_data_min_length': None,
 'tx_padding': 0xCC,                        # Will pad all transmitted CAN messages with byte 0x00.
 'rx_flowcontrol_timeout': 1000,         # Triggers a timeout if a flow control is awaited for more than 1000 milliseconds
 'rx_consecutive_frame_timeout': 10000,   # Triggers a timeout if a consecutive frame is awaited for more than 1000 milliseconds
 'max_frame_size': 4095,                 # Limit the size of receive frame.
 'can_fd': True,                        # Does not set the can_fd flag on the output CAN messages
 'bitrate_switch': True,                # Does not set the bitrate_switch flag on the output CAN messages
 'rate_limit_enable': False,             # Disable the rate limiter
 'rate_limit_max_bitrate': 1000000,      # Ignored when rate_limit_enable=False. Sets the max bitrate when rate_limit_enable=True
 'rate_limit_window_size': 0.2,          # Ignored when rate_limit_enable=False. Sets the averaging window size for bitrate calculation when rate_limit_enable=True
 'listen_mode': False,                   # Does not use the listen_mode which prevent transmission.
}

def setup_can_interface():
    try:
        # Set up the CAN interface for Vector hardware
        bus = can.Bus(interface="vector", channel= 0, app_name="CANoe")
        return bus
    except (can.CanError, OSError) as e:
        print(f"Error setting up CAN interface: {e}")
        return None
        return None

def setup_isotp_stack_f2(bus):
    try:
        # Set up the ISO-TP layer over the CAN interface
        tp_addr = isotp.Address(isotp.AddressingMode.Normal_29bits, txid=0x18DA9BF2, rxid=0x18DAF29B)
        stack = isotp.CanStack(bus, address=tp_addr, params=isotp_params)
        print(stack.params.bitrate_switch)
        print("done")
        return stack
    except Exception as e:
        print(f"Error setting up ISO-TP stack: {e}")
        return None

def send_uds_request(stack, data):
    try:
        stack.send(data)
        print(f"UDS request sent: {data}")
    except (isotp.CanError, OSError) as e:
        print(f"Error sending UDS request: {e}")

def receive_uds_response(stack, timeout=1.0):
    response = None
    timeout_time = time.time() + timeout
    while time.time() < timeout_time:
        stack.process()
        if stack.available():
            response = stack.recv()
            break
        time.sleep(0.01)  # Sleep to prevent high CPU usage
    return response

def main():
    bus = setup_can_interface()
    if bus is None:
        return
    
    
    stack_f2 = setup_isotp_stack_f2(bus)

    print("F2- Physical")
    if stack_f2 is None:
        return
    
    # Example UDS Diagnostic Session Control request (0x10 0x01)
    uds_request_data = [0x10, 0x01]
    
    # Send the UDS request
    send_uds_request(stack_f2, uds_request_data)
    
    # Wait for the response
    response = receive_uds_response(stack_f2)
    
    if response:
        print(f"UDS response received: {response}")
    else:
        print("No response received within the timeout period")

if __name__ == "__main__":
    main()
```

Thank you :)

hartkopp, 2024-07-01:
A CAN frame is never filtered on the value of the BRS bit.
Are you sure, you are creating CAN FD frames in both cases?
Because a CAN ID 0x123 with Classical CAN is a different thing than CAN ID 0x123 with CAN FD.
Please add a tab in your Vector setup to display the CAN frame type to see whether it is CAN CC or CAN FD.

karthik-bammidi, 2024-07-02:
Hi @hartkopp 

Thanks for quick response, 
I didn't find the field in vector to display CAN frame type 

[image]

pylessard, 2024-07-02:
@hartkopp is probably right. 
You should normally be able to add the column by right clicking a column header.

To enable CAN FD in Python, pass `fd=True` to your bus constructor

https://python-can.readthedocs.io/en/stable/interfaces/vector.html#vectorbus

https://github.com/hardbyte/python-can/blob/main/can/interfaces/vector/canlib.py#L99C9-L99C11

karthik-bammidi, 2024-07-02:
Thank you :)

Issue got fixed

closed 2024-07-02 by karthik-bammidi